guysoft/OctoPi #499: No /dev/video0

chilinski, 2018-03-07:
I installed OctoPi (Linux octopi 4.9.28+ #998 Mon May 15 16:50:35 BST 2017 armv6l GNU/Linux) on a Pi B (the one with only two USB ports). It has been running great with one small issue. There is no webcam/streaming. At one point I received a message about /dev/video0 not being found. And sure enough, there is no /dev/video0. There's a /dev/video1. My webcam is a Logitech C270. I found tonight that there is a video0 reference in webcamd. After my current print job is finished, I will change it to video1 and reboot and see what happens.

guysoft, 2018-03-07:
Please try the latest nightly and report. This fix was just merged in to the devel branch 8 days ago: https://github.com/guysoft/OctoPi/pull/492

chilinski, 2018-03-07:
This is interesting. In webcamd, I changed /dev/video0 to /dev/video1 in two places. Rebooted. I now had a /dev/video0 and not /dev/video1. The camera was not found. I changed it back to /dev/video1 in webcamd and rebooted and now I had /dev/video1 but not /dev/video0. I was greatly amused.

guysoft, 2018-03-07:
Please try the nightly which should solve any of those cases. No need to reinvent the webcamd.

chilinski, 2018-03-08:
Well, this is incredibly amusing. I downloaded the latest nightly and burned it and put it in my Pi B2. I booted. No octoprint running. So I tried to start it. It didn't start. Check the logs, you say. Well the only thing in the ~/.octoprint directory is one config.yaml file. There is no log directory. There are no directories at all in .octoprint. I re-downloaded the zip file, uncompressed it again with 7zip, reloaded it on the card and it's the same. Also wired networking refuses to work as long as the camera is plugged in.

guysoft, 2018-03-08:
There was another bug, which was fixed. Will need another nightly.
Nightlies are not the most stable thing in the universe:
https://github.com/guysoft/OctoPi/commit/215b2edef0e7d20ac27512bc4d8b4bd8da9db57b

chilinski, 2018-03-09:
I am abandoning efforts to get the webcam working. I believe it might be an issue with using the old Pi. If I could get the Pi to see the camera, I wasn't seeing the serial port. If I plugged in the camera after it was booted and running, I would lose ethernet. I thought it might be a power issue so I tried a wired hub and that didn't help at all. I simply cannot get ethernet (either using the built-in or a usb dongle), serial port and camera all to work at the same time. I tried different combinations of where stuff was plugged in suspecting possibly a flaky usb port, but that didn't change anything. I think if I get to the point where I really need the camera to work, I'll buy a Pi 3.

guysoft, 2019-04-23:
Closing doe to no more input form original reporter.

closed 2019-04-23 by guysoft