guysoft/OctoPi #502: Octopi wlan0 always gets APIPA IP address (169.254)

whodatwh0dat, 2018-03-08:
Hey all,

My octopi connects to the WiFi router but won't get an IP address via DHCP.
Funny thing is, I can SSH to it via it's hostname (octopi.local), which resolves to an APIPA IP, but not via it's APIPA IP directly.

I tried to set it up like a regular raspbian as well as tried connecting it to different WiFi routers, but same problem occurs. I also re-imaged octopi.

Could you please help me out? I never had such issue with RPi.

Thanks

guysoft, 2018-03-08:
Does it do the same thing with plain Raspbian?

whodatwh0dat, 2018-03-08:
@guysoft no, I always get an IP.

btwood, 2018-04-14:
I believe that CIDR is used by some systems, including windows, if a DHCP server could not be reached.

Try relocating your Pi, and ensuring you've got a good network connection. Try changing your wireless router to "auto" channel selection, and reboot it in order to switch to a wifi channel that isn't as saturated.

Make sure you don't have any other wireless interference going on, such as neighbors, poorly shielded microwaves, dead-zones, other clients. Ultimately a direct line of sight between the client and the router is best. Make sure the wireless signal doesn't have to go through a lot of walls/metal/shielding in order to reach the router.

Lastly, the Zero W uses 802.11n, which means it's going to be 2.5 GHz only. You might be able to get better network performance if you buy an extrernal USB dongle that does 5 GHz if your router supports it and your printer isn't too far away from your router.

There are certainly other ways to solve this issue, including trying to create a wireless mesh in your house, but that is probably well beyond the scope of cost and reason for most people.

Another option is to add an external antenna, but that requires some soldering and some understanding of the board schematic. In theory a fixed wire could work just as well, but I honestly don't know.
https://www.briandorey.com/post/Raspberry-Pi-Zero-W-external-antenna-mod

guysoft, 2019-04-23:
Strange actions by reporter, tried to destroy his report. Closing because I don't know why.

closed 2019-04-23 by guysoft