guysoft/OctoPi #504: Octopi wont bootup

jorksx, 2018-03-14:
Trying to install octopi on rasbperry pi1 
Downloaded octopi-jessie-lite-0.14.0.img and flashed it with win32 disk imager

When I insert the card into rasbperry and plug in usb to power it up, nothing happens. Red led is solid.
Tried to un comment the wifi setting so I can use wifi dongle but my router does not give IP to rasbperry.
So I think that rasbperry wont boot.
What did I do wrong?
Tried with ethernet cable, still same thing. 

rasbperry works and SD card works, I tried with linux and everything worked first time.
So the problem is with octopi and with my rasbperry.

guysoft, 2018-03-14:
Please use a screen to see if the pi boots

jorksx, 2018-03-14:
oh yeah. forgot to mention that ive tried plugging in keyboards and screen. I see No signal. downloaded that noobs os lite and worked right away

guysoft, 2018-03-14:
Are you using the same SD? 
Did you extract the image amd flash it?

If you see no colorful screen on boot its a hardware issue.

jorksx, 2018-03-14:
i have tried tree diffrent SD card  they all work with linux and raspberry. when flashing octo it does not work anymore. I see only black screen. 
how can it be hardware issue if all linuxes work fine?

guysoft, 2018-03-14:
If you don't see the rainbow image at boot it means the kernel on the SD card image is not recognized. 
OctoPi is built on top of raspbian, which maintain the kernel part of the boot.  If you can't see that then the scope if your issue is beyond what OctoPi maintains.
Therefore there is nothing in this code base that can fix such an issue without heavily modifying the boot image.

However,  afrer over 500 issue here, its from my experience that an issue like yours (aka a single issue with no other reporters that nothing works before the machine even boota) are usually small trivial with something like
* the image is not flashed correctly 
* a fauly power supply not giving 2A current
* bad usb cable
* unusual hdmi screen that doesn't work with the pi
* bad download
* flash the zip or copying it instead of flashing

Ill change my assumption if there are more reporters

John-Mc, 2018-03-14:
@jorksx  Is your screen connected and powered in the same manner you used with other Raspian installations? Did you have to do any special set-up (i.e. edit any files on the SD Card) to get your screen to work?)

Your screen needs to be plugged in and ready to go BEFORE you power up the Pi. If you have done that, power up the pi and watch the screen the whole time. Early on in th epower up process, you should see a rainbow colored screen for a few seconds (this is what Guy was referring to with the "colorful screen").

Were your other installations all NOOBS-based, or were some of them a standard Raspian or Raspian-Lite install (The latter has a different installation process than doing it via NOOBS).

jorksx, 2018-12-25:
I fixed mine with 32gb SD card

boelle, 2018-12-25:
the sd card works fine, right now writing the raspbian-stretch-lite image and trying to install from scratch

jorksx, 2018-12-25:
Mine was same. I coule run ubuntu fine from it. No problems. But for some reason with octopi, it did not work

boelle, 2018-12-25:
i did not mean octopi but the "standard" image from raspberry pi org

if that works i will install octoprint the manual way as here: https://discourse.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337

boelle, 2018-12-25:
I wonder if the  Pi 3 B is supported? i was cheap and did not get the + model

jorksx, 2018-12-25:
Yes  I ran rasbian and ubuntu without any problems on my raspberry 1. But when I tried octopi i got errors written up when I started this post. Then I just bought new SD card and octopi started working fine

boelle, 2018-12-25:
hmm... seems i have to do the same... not even rasbian will run anymore, 

not even the rainbow image comes up, will try same sd card in another pi and see if its the same there

jorksx, 2018-12-25:
SD cards are cheap. It was wierd for me because rasbian and ubuntu ran fine but octoprint not

John-Mc, 2018-12-25:
Pi 3B is supported by OctoPi 0.15.1 (the latest stable release). I am currently running OctoPi 0.15.1 on my Pi 3B with no problems (as are hundreds of other people).  The Pi 3A+ is newer than the 3B+ and I do not believe is supported at this time.

As the problem you are experiencing is not a bug, but more likely a set-up problem, I'd suggest that the Discourse OctoPrint link you posted is a better place to troubleshoot this. I've replied to you on the thread you linked.

boelle, 2018-12-25:
kingston cards are not, and sadly shops are closed, stupid xmas :-/

boelle, 2018-12-25:
@John-Mc you are right, just need tips and pointers  :-D