guysoft/OctoPi #505: Raspberry Pi B+ and new 2018-03-13 image

guysoft, 2018-03-14:
Happy Pi day all,
New image, so nightly out soon with it so you guys can test and report here (living in Israel I can expect a new Pi only in about a month.

Image would be posted here when it finishes building.

```
2018-03-13:
  * Raspberry Pi 3 B+ support
  * WiFi is disabled until wireless regulatory domain is set (Pi 3 B+ only)
    - The domain can be done through 'Raspberry Pi Configuration' (rc_gui),
      'raspi-config' or by setting 'country=' to an appropriate ISO 3166
      alpha2 country code in /etc/wpa_supplicant/wpa_supplicant.conf.
  * Default wireless regulatory domain is now unset
  * Added support to desktop for different screen sizes and resolutions,
    including multiple preset options in Appearance Settings and pixel doubling
    option in Raspberry Pi Configuration
  * Version 2.1.16 of Thonny included
  * Version 29.0.0.113 of Adobe PepperFlash player included
  * Version 1.2.post1 of Pygame Zero included
  * Bluetooth plugin now supports connection to Bluetooth LE HID devices
  * Network plugin now indicates 5G-compatible APs
  * Latest changes to Bluez ALSA service merged
    - service now started on CLI boot as well as GUI boot
  * Latest changes to dhcpcd networking plugin merged
  * Improved support for running on pi-top devices
  * Small design changes to PiX theme and icons
  * Bug fix - hide spurious window resize handles
  * Bug fix - Scratch 2 remote GPIO state block now works correctly
  * Updated WiFi Firmware
    - brcmfmac43455-sdio 7.45.154
    - brcmfmac43430-sdio 7.45.98.38
```

guysoft, 2018-03-14:
Here is the nightly:
https://storage.googleapis.com/octoprint/2018-03-14_2018-03-13-octopi-stretch-lite-0.15.0.zip
md5: 67f42949897e152079c276d4811989c5

Follow for new nighties at: http://gnethomelinux.com/OctoPi/nightly/

guysoft, 2018-03-14:
[image]
Had a test run today (I don't own this Pi, so only one boot test, can't test anynore). Seems to boot, no wifi. So anyone that manages to get the wifi working please report here.

guysoft, 2018-03-15:
New nightly, I can't test it so waiting for people to confirm it works, or not:

https://storage.googleapis.com/octoprint/2018-03-15_2018-03-13-octopi-stretch-lite-0.15.0.zip

md5:
b0b3449fe343d10ebe88c04df46e27a1

CJeffyB, 2018-03-17:
Would it be helpful to test out the nightly on a non-B+ Pi 3?  I haven't been able to find any place that has the new version in stores, so I'm still running the original Pi 3.

How much testing does a new version need prior to being considered a candidate for number bump?

guysoft, 2018-03-17:
@CJeffyB yes, any kind of test would help. Because is a new image. 
Testing on the older devices than the 3B+ does not help to confirm the wifi issue was fixed, but it will confirm that the Rpi foundation didn't break anything. That also includes RaspberryPi zero. They made general changes to the kernel.

How much testing? Well enough that we feel safe to release, but not too much to get make people waiting. Depends really on how many people volunteer to test. We have no rigorous definition, we are all volunteers (except @foosel who is living of crowdfunding).

John-Mc, 2018-03-18:
BTW, you can also set `country=` in /boot/octopi-wpa-supplicant.txt, can't you? (So it could be done while SD card is in your computer, before booting on the Pi.)

stevendpclark, 2018-03-18:
I can confirm the wifi is fixed with the attached nightly image. My 3B+ boots up fine and connects properly to my WPA AP.

kirknorris, 2018-03-19:
Stevendpclark, how did you get it to work? i have followed the connection instructions but it just wont connect to my wifi network?

stevendpclark, 2018-03-19:
@kirknorris Nothing special, I flashed a new SD card with the attached image using Etcher, setup boot/octopi-wpa-supplicant.txt settings such as 

```
network={
  ssid="XXXXX"
  psk="ZZZZZZZ"
}

country=CA
```

Booted up and it connected directly to my Ubiquiti AC-Pro AP with no additional changes. This didn't work with the older octopi image that was running within my original Raspberry Pi 3B (Non plus version).

kirknorris, 2018-03-19:
@stephendpclark the only difference I can see is that I have the spaces still in the country code, ill try it now and see how it goes!

kirknorris, 2018-03-19:
@stevendpclark no go :( i thought it may be my router but i have tried with my mobile as a hotspot also and no connection! im stumped now, i must be missing something really obvious

stevendpclark, 2018-03-19:
I activated the wireless on my internet providers device and the pi successfully connected to that AP as well, so I think it must be something on your end. Note that it didn't properly connect at first as I hadn't matched the capitalization of the wireless name perfectly, so perhaps check that on your end. 

Also I performed a apt-get update && apt-get dist-upgrade on the image and it now connects to my main AP as a 5Ghz

guysoft, 2018-03-19:
Also make sure to set your router and pi to the same country. If your router has a firmware from a different country set it to that.

joshcarter, 2018-03-20:
@guysoft I can also confirm that your 2018-03-15 nightly image works on Pi 3 B+. Just set it up from scratch and tested USB printer connection, camera, and PSU control via GPIO. Connected via wifi to my 802.11ac network, set up per comments in this thread. Thanks all!

guysoft, 2018-03-20:
@joshcarter pi pr usb camera? Getting reports that the camera might be doing something:
https://discourse.octoprint.org/t/streaming-hangs-on-raspi-camera-v2-rpi-3-b/399?u=foosel meanwhile people are encountering funny issues with the webcam on the plus

rrauenza, 2018-03-20:
Hi -- I've been referred back here from the the discourse site -- I'll copy the text for reference - I'm using a 3 B+ and when I started this,  2018-03-19_2018-03-13-octopi-stretch-lite-0.15.0.zip was the newest thing out:

**What is the problem?**

When using the streaming URL, I get no output in the browser .. it just seems to wait indefinitely.

Snapshots work.  raspivid and raspistill work.

I also occasionally am getting kernel stack traces on the rpi if I let the browser sit for long enough.  I don't think this is the root cause -- this may be an after effect:

```
[  991.665459] Unable to handle kernel NULL pointer dereference at virtual address 00000028
[  991.671783] pgd = 8d5ac000
[  991.677693] [00000028] *pgd=2d8f8835, *pte=00000000, *ppte=00000000
[  991.683974] Internal error: Oops: 17 [#1] SMP ARM
[  991.690023] Modules linked in: xt_tcpudp iptable_mangle xt_DSCP cmac bnep hci_uart bluetooth ecdh_generic evdev brcmfmac brcmutil cfg80211 snd_bcm2835(C) rfkill snd_pcm snd_timer snd fixed uio_pdrv_genirq uio ip_tables x_tables ipv6
[  991.703799] CPU: 1 PID: 634 Comm: mjpg_streamer Tainted: G         C      4.14.27-v7+ #1100
[  991.710696] Hardware name: BCM2835
[  991.717548] task: 8d5a6900 task.stack: 8c860000
[  991.724421] PC is at tcp_push+0x44/0x148
[  991.731310] LR is at tcp_sendmsg_locked+0x308/0xdfc
[  991.738236] pc : [<806ce054>]    lr : [<806d1620>]    psr: 40000013
[  991.745187] sp : 8c861d78  ip : 8cc5267c  fp : 8c861d9c
[  991.752241] r10: 8cf9ea80  r9 : ffffffe0  r8 : 8cc5267c
[  991.759277] r7 : 00000000  r6 : 0000ffcb  r5 : 00000000  r4 : 8cc52580
[  991.766522] r3 : 00000000  r2 : 0000ffcb  r1 : 00000000  r0 : 8d59c900
[  991.773684] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  991.777192] Control: 10c5383d  Table: 0d5ac06a  DAC: 00000055
[  991.780479] Process mjpg_streamer (pid: 634, stack limit = 0x8c860210)
```

**What did you already try to solve it?**

I started with 2018-03-19_2018-03-13-octopi-stretch-lite-0.15.0.zip that supports the b+.

I originally had issues detecting a camera so started trying some various things based on suggestions I found in google:

I changed this:

```
$ grep gpu /boot/config.txt 
gpu_mem=256
```
I also explicitly set the camera to raspi:

```
$ grep -v ^# /boot/octopi.txt 
camera="raspi"
```

I also did an apt-get update then apt-get upgrade.  I ran raspi-config to make sure the camera was enabled.  I also ran rpi-update to update firmware.

At this point, I was able to get raspistill and raspivid to work (if I stopped the webcamd - some of the steps above were done because I didn't realize I needed to stop webcamd to get raspstill to work.)

I am able to get a static image through http://10.13.54.180:8080/?action=snapshot

A streaming image from http://10.13.54.180:8080/?action=stream just sort of hangs in the browser.  If I kill webcamd, I end up with a static image on the browser.  If I curl the same URL I do get a constant stream of bits.

Here's what is actually running:

```

├─webcamd,431 /root/bin/webcamd
  │   └─mjpg_streamer,448 -o output_http.so -w ./www-octopi -n -i input_raspicam.so -fps 10
  │       ├─{HCEC Notify},454
  │       ├─{HDispmanx Notif},452
  │       ├─{HTV Notify},453
  │       ├─{VCHIQ completio},451
  │       ├─{mjpg_streamer},449
  │       ├─{mjpg_streamer},450
  │       ├─{mjpg_streamer},633
  │       ├─{mjpg_streamer},927
  │       ├─{vc.ril.camera},455
  │       ├─{vc.ril.image_en},467
  │       └─{vc.ril.video_re},456
```

The log...

```
Starting up webcamDaemon...

--- Configuration: ----------------------------
camera:        raspi
usb options:   -r 640x480 -f 10
raspi options: -fps 10
http options:  -w ./www-octopi -n
-----------------------------------------------

<13>Mar 20 04:41:42 pi: Starting Raspberry Pi camera
Running ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_raspicam.so -fps 10
MJPG Streamer Version: git rev: 8cc9d22c1e79905d529a248ccf05bbf0625e0bf3
 i: fps.............: 10
 i: resolution........: 640 x 480
 i: camera parameters..............:

Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
 o: www-folder-path......: ./www-octopi/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: disabled
 i: Starting Camera
Encoder Buffer Size 81920
```

**Additional information about your setup** (OctoPrint version, OctoPi version, printer, firmware, `octoprint.log`, `serial.log` or output on terminal tab, ...)

Not connected to a printer.  OctoPrint 1.3.6 running on OctoPi 0.15.0.

kirknorris, 2018-03-20:
Ok guys, thanks for the advice! I have managed to solve the problem I was having with the pi not connecting and the problem was my SD card, i replaced it with a reputable class 10 and all seems good so far! I just need to find a camera that's good for it now 👍 thanks again guys and especially @stephendpclark

joshcarter, 2018-03-20:
@guysoft I'm using the Pi camera (version 2), not USB. It came right up with zero issues.

rrauenza, 2018-03-21:
I've redone my setup, but documented it this time.  It still isn't working, same problem.  I didn't change any config this time, and also didn't apply any external updates.  Here are my notes:

Using Raspberry Pi 3 B+
Using Raspberry Pi Camera v2.1

Downloading latest 0.15 Nightly...

	https://storage.googleapis.com/octoprint/2018-03-20_2018-03-13-octopi-stretch-lite-0.15.0.zip
	Extract img from zip

Using Etcher to burn into 32GB SandDisk Extreme

Edit octopi-wpa-supplicant.txt

Hooked up HDMI...

Power on, Booting... (auto Resized root filesystem, then auto rebooted again)

Logged in via HDMI console:

$ vcgencmd get_camera
supported=1 detected=1

$ sudo service webcamd stop

$ raspistill -o /tmp/foo.img 
<works, shows image>
$ raspistream -o /tmp/foo.img 
<works, shows video>

$ sudo service webcamd start

Trying web.

Setting up accounts, etc., ...

Going to 'Control Panel' and it just says 'Webcam stream loading'

Logging in over ssh:

$ pstree -an 
[...]

  └─webcamd /root/bin/webcamd
      └─mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_raspicam.so -fps 10
          ├─4*[{mjpg_streamer}]
          ├─{VCHIQ completio}
          ├─{HDispmanx Notif}
          ├─{HTV Notify}
          ├─{HCEC Notify}
          ├─{vc.ril.camera}
          ├─{vc.ril.video_re}
          └─{vc.ril.image_en}
		  
Rebooting to try once more ....

Logging into web console, looking at Control tab:

Nope, same problem.

rrauenza, 2018-03-21:
Snapshots work.  Just not streaming.   If I use curl to poke the stream url, I get content -- the first of it has this content:

```
--boundarydonotcross
Content-Type: image/jpeg
Content-Length: 155471
X-Timestamp: 0.000000

<FF><D8><FF><E1>^@<DE>Exif^@^@MM^@*^@^@^@^H^@^H^A^@^@^D^@^@^@^A^@^@^B<80>^A^A^@^D^@^@^@^A^@^@^A<E0>^A^P^@^B^@^@^@       ^@^@^@n^A^Z^@^E^@^@^@^A^@^@^@x^AESC^@^E^@^@^@^A^@^@^@<80>^A(^@^C^@^@^@^A^@^B^@^@^B^S^@^C^@^@^@^A^@^A^@^@
<87>i^@^D^@^@^@^A^@^@^@<88>^@^@^@^@Broadcom^@^@^@^@^@H^@^@^@^A^@^@^@H^@^@^@^A^@^F<90>^@^@^G^@^@^@^D0220<91>^A^@^G^@^@^@^D^A^B^C^@<A0>^@^@^G^@^@^@^D0100<A0>^A^@^C^@^@^@^A^@^A^@^@<A0>^B^@^D^@^@^@^A^@^@^B<80><A0>^C^@^D^@^@^@^A^@
^@^A<E0>^@^@^@^@<FF><DB>^@<84>^@^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A^A

```

rrauenza, 2018-03-21:
Ok, well, after researching mjpeg streams, and verifying that the output of the curl has multiple headers of 
```
--boundarydonotcross
Content-Type: image/jpeg
Content-Length: xxxxxx
X-Timestamp: 0.000000
``` 

I decided I must be crazy because the format looks legit.  So I'm trying from a different computer, but still using Chrome and it works.  ...I wonder.... the browser I was using has qustodio installed, and while not actively filtering for *my* windows account, does seem to hijack all web traffic for all potential users.

I didn't change anything since last night except to install lsof.

Now, my only other problem is it doesn't seem to communicate with my Prusa mk3, but I think that's because it's on a UPS truck in a box ;)

hoberg63, 2018-03-21:
Today got. my 3B+ install build 2018-03-20 Boot ok, Connection to Anycubic ok, Wifi ok, Webinterface ok, next testing Print. Friday i got my IR camera.
Will then Test everything
Thanks for the build

hoofrog, 2018-03-24:
Disclaimer: I am new to Raspberry Pi and OctoPi. I borrowed a Pi 3B and had no issues testing out OctoPi and deciding I definitely needed it for my setup. I ordered a 3B+ and this morning downloaded the latest nightly build to install. The only issue I have at the moment is connecting to the Pi via SSH. I keep getting connection denied.

vertigo235, 2018-03-24:
Just wanted to report that I have been using the 2018-03-15_2018-03-13-octopi-stretch-lite-0.15.0 for several days now and it is working just as expected, no printing issues, and USB webcam works fine as well.

jeffeb3, 2018-03-27:
@guysoft Raspberry pi foundation should really ship you one of each pi as they come out, considering the number of octopi's out in the world thanks to this repo.

hoberg63, 2018-03-27:
Now done a lot of Prints, Same Build, added now IR Pi 5mp Camera Modul, no issue, Plug and Play.
WhatsApp new in the earlier Builds ?

Rasberry PI 3 B+, PI IR Camera

guysoft, 2018-03-27:
Thanks @jeffeb3  <3
Just heads up that although nightly looks stable aand usable for 3B+ (minus some camera reports) we are waiting for OctoPrint 1.3.7 to release.

sfsdfd, 2018-03-29:
>   * WiFi is disabled until wireless regulatory domain is set (Pi 3 B+ only)

This one change cost me 3-4 hours of fiddling with WiFI settings on a Pi 3B+.

I've configured about a dozen Raspberry Pis over the past year using a well-established process, so it surprised me when the new 3B+ didn't appear on the network after following the exact same set of steps. I was about to return it as having a dead WiFi chip when I noticed the rfkill line in journalctl.

More importantly: rfkill appears to get tripped if the country code isn't specified **even once** - and then persists across reboots **even if you later specify the country code**. The only way to restore functionality, after adding the country code, is to disable rfkill manually.

I can speculate as to why this change was made - i.e., that 802.11ac isn't fully licensed in all countries. However, inserting this new requirement in this way is really problematic. There's nothing in the Raspberry Pi 3B+ pack-in materials about this. There's no obvious page on raspberrypi.org that's been indexed with a notification like: "RASPBERRY PI 3B+ OWNERS EXPERIENCING WIFI PROBLEMS" with a description and a solution. I ran *a lot* of searches to diagnose the issue, and just found random chatter without a clear indication of the issue.

This is a serious problem that will cause a lot of headaches until it's a well-known issue with a well-known resolution.

John-Mc, 2018-03-29:
The recent nightly releases have the WiFi country set to GB by default in /boot/octopi-wpa-supplicant.txt (see near the bottom of the file), with an explanation that you should "Uncomment the country your Pi is in to activate Wifi in RaspberryPi 3 B+ and above" (to change it to your own country).  I know the 2018-03-25 nightly defaulted to GB. So at least there is a default in OctoPi that avoids the issue.  

I had not realized that rfkill persisted, even if you later specified a country. That was an "interesting" choice (I presume by the Raspberry Pi / Raspian folks?)

foosel, 2018-03-30:
@sfsdfd

This is a serious problem that will cause a lot of headaches until it's a well-known issue with a well-known resolution.

Agreed. Which is why when this new behaviour was discovered in the Raspbian base image Guysoft and me immediately looked into how to revert the OctoPi image to the former behaviour. Which got into effect on the nightly builds less than 24h after release of the 3+.

I suggest you voice your (rightful) complaints towards the Raspberry Pi Foundation that introduced that change, not the people who immediately jumped on to roll it back in a customized image :)

sfsdfd, 2018-03-30:
Oh, I get that, Gina. My irritation certainly wasn't directed at the OctoPrint team!

My writeup was intended to describe how this problem is manifesting in the field - I'd like to spread the word both to RPi developers and, even more importantly, other users who are encountering the same behavior. I've already posted on Reddit and the official RPi Forum as well, though the sheer volume of those forums may cause those posts to get lost.

Notably - when I started searching for threads about RPi WiFi issues, this thread was one of the first to arise - completely incidental to OctoPrint. Hence why I posted it here.

guysoft, 2018-03-30:
Just saying, I posted this said, you can voice it here too, there are RaspberryPi engeneers on that thread which answered us:

> FYI,
> This is how ready we can get when it comes to new releases. We know that when they ship there will be bugs.

https://www.raspberrypi.org/forums/viewtopic.php?t=187797

jesasi, 2018-04-02:
Dunno is this helping someone, but i had fully working octopi image running on my pi3.

I did put that same microsd card to my pi3+, managed to get it up and running, except wlan0 was nowhere to seen.

After couple hours i found that " /lib/firmware/brcm/brcmfmac43455-sdio.bin " was missing, it should come with "firmware-brcm80211", but for reason what i don't know, it did not come there.

I manually copied "brcmfmac43455-sdio.bin" file that folder, reboot and wlan was working fine...

guysoft, 2018-04-02:
@jesasi did you set the country?  I know that 3B+ requires tje ccountry to be set toto work. Perhaps the new image plays with the firmware?

daboi, 2018-04-02:
I installed 30.03.2018 nightly install to 3B+ and after setting wifi password in text file it connected without any problems to my 2.4g network.

Emre

jesasi, 2018-04-02:
@guysoft Yes, that was first what i did, but did not help then.

christopherleegould, 2018-04-11:
I can confirm Wifi working correctly on a Pi 3 B+. I followed the wpa_supplicant directions, but forgot to uncomment the #network and #} lines first. I suspect it would have worked right out of the box if I had used raspi-config first instead of doing it manually. I didn't even catch my mistake until raspi-config said it couldn't communicate with the wpa supplicant, and I knew I had butchered the file.

Drmarvin, 2018-04-12:
I can also confirm WiFi working with Pi3 B+ with the nightlies and the change to the wpa_supplicant file. I am having an issue with it, however. Tested bed heating after connecting, was able to set the bed heat and the target heat level showed on the printer but then it never actually heated and the printer errored with a communication to heat bed error but I was able to do it manually.

juan9999, 2018-04-13:
I installed 2018-04-11_2018-03-13-octopi-stretch-lite-0.15.0.zip

WiFi works, but I can't connect to my CR-10S.  

When I try to connect to my new CR-10S, I get the following error:

Error reported by printer
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: MINTEMP triggered, system stopped! Heater_ID: 0

guysoft, 2018-04-13:
@juan9999 that's a legitimate error, fix your thermisor and/or its connector.

juan9999, 2018-04-13:
@guysoft It's actually printing fine right now from the SD card...  And the correct temps are showing on the CR-10 display...

guysoft, 2018-04-13:
Report it to the OctoPrint issue tracker, unrelated to OctoPi https://github.com/foosel/OctoPrint/issues

foosel, 2018-04-13:
No, don't open a ticket for that. First of all you already opened a thread on the community forum unless I'm mistaken, and secondly: This is your firmware reporting a serious error. I have no idea why it works when you print from SD and not via serial, but in any case, this is not something that OctoPrint (or OctoPi) can solve. If your firmware reports catastrophic heater failures on the serial interface, it's not possible to use that interface. Figure out why your printer does that and fix the underlying issue.

Broxie, 2018-04-17:
I setup nightly on Pi 3b+ using nightly (https://storage.googleapis.com/octoprint/2018-03-14_2018-03-13-octopi-stretch-lite-0.15.0.zip).  SD card is 32GB Samsung EVO Plus class 10.

The camera I am using is one from the Pi Zero W with an adapter into the camera port.

I edited the octopi-wpa-supplicant.txt

network={
  ssid="XXXXX"
  psk="ZZZZZZZ"
}

country=GB

* I did not need to un-hash the country code it was already set to my location.

The camera, controls and usb loading all working and ran my first print jpb on the Anet A8 which has been flashed with Marlin.

I do not have a cura-ini file so I sliced the item in Cura and uploaded the gcode to Octopi. It has successfully printed  the object.

juan9999, 2018-04-18:
For others having this issue:

Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: MINTEMP triggered, system stopped! Heater_ID: 0 Problem 

For me it was SOLVED by flashing Marlin.

guysoft, 2018-04-23:
Rpi 3B+ works. Rpi released fixes for stuff, please see #516

closed 2018-04-23 by guysoft