guysoft/OctoPi #507: Password reset issue between direct reset and via ssh

suvinks, 2018-03-23:
During my first attempt, i reset the password on the pi from "raspberry" to my own password. This I did directly on the raspberry pi login (did not ssh). Once that was done successfully, I tried to login via ssh with the new password. I got "Access denied". Not even the "raspberry" password worked. 
Next time reinstalled and I reset the password by logging through ssh using "raspberry". At this point am able to access directly and via ssh with the password "raspberry". 
The Moment I changed the password via ssh, I can no longer connect directly to pi (not with new or default password). It gives me access denied. Am able to login through ssh with the new password.

Are you aware of this issue? Any resolution?

guysoft, 2018-03-23:
Do you get the same behavior with raspnian?

suvinks, 2018-04-10:
i didnt try in raspian. i downloaded your image and tried on it. Should i try in generic raspian on the pi?

guysoft, 2018-04-11:
You are describing that part of the system that is part of raspbian is failing. Apart from enabling ssh we don't really change anything.
However it might be some strange connection or something else that iis going on.
Testing with raspbian is a way to rule out the most likely cause which is that you made a misstake somewhere (we all do), and also means its sometimes we are changing in OctoPi.