guysoft/OctoPi #508: Wireless network power management by default on

bipsendk, 2018-03-28:
Tried to install the nightly 2018-03-13-octopi-stretch-lite-0.15.0.img - and after a startup, I did a iwconfig to check the wireless LAN:

wlan0     IEEE 802.11  ESSID:"xxxxxx"
          Mode:Managed  Frequency:2.422 GHz  Access Point: xx:xx:xx:xx:xx:xx
          Bit Rate=72.2 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          **Power Management:on**
          Link Quality=65/70  Signal level=-45 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

I think, that power management of the WiFi by default should be switched off.... Maybe something, that can be included in one of the config files under /boot ?

SamGurdus1, 2019-09-16:
I know this is an old issue, but I'm running into it now. How do I manually disable power management? All the info I can find online mentions files that are blank when I nano them.

guysoft, 2019-09-16:
@SamGurdus1 What version of OctoPi?

jotapdiez, 2019-10-07:
@guysoft In my case 0.16.0

pandel, 2021-09-07:
@guysoft I have the same problem with the newest OctoPi release.

To solve this, please
- add script `powersave_off` to `/etc/network/if-up.d/`
- make it executable `chmod +x /etc/network/if-up.d/powersave_off`
- with content:

```
#!/bin/sh

set -e

# Don't bother for loopback
if [ "$IFACE" = lo ]; then
        exit 0
fi

# Only run from ifup.
if [ "$MODE" != start ]; then
        exit 0
fi

# Only run once
if [ "$ADDRFAM" != meta ]; then
        exit 0
fi

/usr/sbin/iw dev wlan0 set power_save off

exit 0
```

This solved the Powermanagement problem permanently.

And generally, keep up the good work and stay safe!

Regards,
Holger

guysoft, 2021-09-09:
Hey there is a new nightly builds out with a fix for this, if anyone want test test and confirm that would help a lot!

pandel, 2021-09-09:
@guysoft Many thanks! ATM, I have only one printer to test things on and it is currently in a 18 hour print...

TheRaf974, 2022-12-05:
Hi, 

I had the same problem with Power Management enabled.

Yesterday, I tried creating a [service to turned it off at startup](https://raspberrypi.stackexchange.com/questions/96606/make-iw-wlan0-set-power-save-off-permanent) and it seems to work fine.

I'll test [pandel's solution today](https://github.com/guysoft/OctoPi/issues/508#issuecomment-914501732) and give you my feedback.

guysoft, 2022-12-05:
Cool, it might be good to add it to the network module at
https://github.com/guysoft/CustomPiOS/tree/devel/src/modules/network

TheRaf974, 2022-12-05:
Ok, 
Pandel's solution doesn't work for me. 

At startup, the power management is deactivated. 

> pi@octoprint:~ $ iwconfig
> lo        no wireless extensions.
> 
> wlan0     IEEE 802.11  ESSID:"******"
>           Mode:Managed  Frequency:2.452 GHz  Access Point: B4:B0:24:F1:90:E0
>           Bit Rate=72.2 Mb/s   Tx-Power=31 dBm
>           Retry short limit:7   RTS thr:off   Fragment thr:off
>           Power Management:off
>           Link Quality=70/70  Signal level=-32 dBm
>           Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
>           Tx excessive retries:82  Invalid misc:0   Missed beacon:0

But, the wifi already cut 2 time in 2 hours, so he probably got back on.

I can make a PR which add `wifi_powersave@.service` if you want.

TheRaf974, 2022-12-11:
Hi @guysoft, 

I don't know how to add this method to [CustomPiOS](https://github.com/guysoft/CustomPiOS), since there is already a [service to turn off wifi power management on Intel Edison boards](https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/network/filesystem/etc/systemd/system/disable-wifi-pwr-mgmt.service).
I think the best way is to adapt the service to the hardware (intel or raspberry) but I don't have enough knowledges to do it.

I does a PR to this repo, to fix this issue for Octopi.

PS: I also had to disable WMM option in my router, to avoid wifi disconnections.

guysoft, 2023-05-29:
Hey, we meeged a pr for this. Might be solved now

guysoft, 2023-05-29:
Hey, we merged a pr for this. Might be solved now

oxivanisher, 2023-09-27:
As an update: As of today (sept. 2023) #799 is not yet in current OctoPi releases.

The old way has additional problems, since it tried to do the `iw` stuff also for eth0 interface which leads to a always failing networking systemd unit. `systemd status` was always degraded.
```plaintext
-- Boot 8672099aa8de4a41ad49283ed7e62e3e --
Jul 20 01:04:15 prusamk4 systemd[1]: Starting Raise network interfaces...
Jul 20 01:04:16 prusamk4 ifup[362]: command failed: No such device (-19)
Jul 20 01:04:16 prusamk4 ifup[321]: run-parts: /etc/network/if-up.d/powersave_off exited with return code 237
Jul 20 01:04:16 prusamk4 ifup[254]: ifup: post-up script failed
Jul 20 01:04:16 prusamk4 systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
Jul 20 01:04:16 prusamk4 systemd[1]: networking.service: Failed with result 'exit-code'.
Jul 20 01:04:16 prusamk4 systemd[1]: Failed to start Raise network interfaces.
```

Before I found this thread, I first fixed it like this:
```bash
#!/bin/sh

set -e

# Only run from ifup.
if [ "$MODE" != start ]; then
    exit 0
fi

# Only run once
if [ "$ADDRFAM" != meta ]; then
    exit 0
fi

# Handle different interfaces
case $IFACE in
    wlan0)
        /usr/sbin/iw dev wlan0 set power_save off
        ;;

    # Don't bother for loopback, eth0 or any other interface
    lo | eth0 | *)
        exit 0
        ;;
esac

exit 0
```
This is only here since it might help somebody in the future.

I manually deployed the systemd unit from #799 and it works like a charm.

guysoft, 2023-10-02:
@oxivanisher It was merged after the RC rounds of OctoPi v1.0. I assume we will relesae soon a new version once we see it works with the upcoming Raspberry Pi 5 (not sure how long that will take).

JaneX8, 2024-01-04:
Potentially related to this same issue: https://raspberrypi.stackexchange.com/questions/145548/why-does-wifi-stops-working-randomly-on-a-raspberry-pi-4b-rev1-4-while-the-wlan0.

guysoft, 2024-01-04:
It might be solved already on OctoPi, if anyone can confirm that would be nice :)