guysoft/OctoPi #509: issues with Raspi B+

mtowara, 2018-03-29:
Hi,
just a heads up,
I'm seeing crashes with my brand new Raspi B+,
which leads to the print failing midway and the pi not responding at all.

This seems to be hardware/kernel related and nothing to do with OctoPi/Print.
I'm using the official 2.5A power brick, thus I'm somewhat confident that its not an power issue.

My best guess is that it is this issue:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=208821

guysoft, 2018-04-01:
I guess this blocks #505 , thanks for reporting

guysoft, 2018-04-23:
There is a new image that I think fixes all this stuff. Please see #516 , if there is no news I will close this.

guysoft, 2018-04-30:
Released https://github.com/guysoft/OctoPi/releases/tag/0.15.0

closed 2018-04-30 by guysoft