guysoft/OctoPi #512: Bit of basic shell usage

pbannister, 2018-04-13:
The bit of helpful text emitted by ~/scripts/welcome is currently invoked from **~/.bashrc** rather than from **~/.profile**. As the comments in each reflect, ".profile" is for login shells ... where you want to emit such notices.

I moved this line from **.bashrc** and added to **.profile**:
```
source /home/pi/scripts/welcome
```

The reason for this is that if I were to (say) have scripts that send single-line SSH commands from another computer, I would not want the "Welcome" message in the output. For example:

```
$ ssh pi@octopi-1.local ls -al
total 84
drwxr-xr-x 14 pi   pi   4096 Apr 12 16:08 .
drwxr-xr-x  3 root root 4096 Jun 21  2017 ..
-rw-------  1 pi   pi   2420 Apr 12 16:06 .bash_history
-rw-r--r--  1 pi   pi    220 Jun 21  2017 .bash_logout
-rw-r--r--  1 pi   pi   3513 Apr 12 16:08 .bashrc
drwxr-xr-x  2 pi   pi   4096 Apr 12 16:00 build-Marlin
drwx------  4 pi   pi   4096 Apr 11 22:41 .cache
-rw-------  1 pi   pi     48 Apr 11 22:15 .lesshst
drwx------  5 pi   pi   4096 Apr 11 22:41 .local
drwxr-xr-x 10 pi   pi   4096 Jun 24  2017 mjpg-streamer
drwxr-xr-x 13 pi   pi   4096 Apr 10 12:24 .octoprint
drwxr-xr-x 12 pi   pi   4096 Apr  4 00:15 OctoPrint
drwxr-xr-x  6 pi   pi   4096 Jun 24  2017 oprint
drwxr-xr-x  2 pi   pi   4096 Apr  4 00:20 .pip
-rw-r--r--  1 pi   pi    880 Apr 12 16:08 .profile
drwxr-xr-x 13 pi   pi   4096 Apr 12 15:23 .pyenv
drwxr-xr-x  6 pi   pi   4096 Apr  4 00:16 .python-eggs
-rw-------  1 pi   pi      0 Apr 11 22:31 .python_history
-rw-r--r--  1 pi   pi      6 Apr 12 15:58 .python-version
drwxr-xr-x  2 pi   pi   4096 Apr 12 16:08 scripts
drwx------  2 pi   pi   4096 Apr  4 00:05 .ssh
-rw-------  1 pi   pi   1217 Apr 12 16:08 .viminfo
lrwxrwxrwx  1 pi   pi     33 Apr 11 22:41 .virtualenvs -> /home/pi/.local/share/virtualenvs
```

(Without the change, the welcome message would appear ... not wanted.)

guysoft, 2018-04-13:
I don't understand what you did. If you want to contribute code please make a pull request.

guysoft, 2018-04-15:
Closing if no reply

pbannister, 2018-04-17:
I removed this line from .bashrc and added to .profile:
```
source /home/pi/scripts/welcome
```
The bug is that you might not be aware of the difference. :)

Made this change as I was preparing to build Marlin, so as to update the firmware on the attached printer. Created a virtual (Python/development) environment, and to enter typed:
```
pipenv build-Marlin
```
... and saw the "Welcome" message, as though I had first logged in. WTF? This is not right ... subshells should not display login messages.

When I go to my other printer (w/o the change), and type **bash** then I get:
```
pi@octopi-2:~ $ bash

------------------------------------------------------------------------------
Access OctoPrint from a web browser on your network by navigating to any of:

    http://octopi.local
    http://192.168.86.23

https is also available, with a self-signed certificate.
------------------------------------------------------------------------------
This image comes without a desktop environment installed because it's not 
required for running OctoPrint. If you want a desktop environment you can 
install it via

    sudo /home/pi/scripts/install-desktop
------------------------------------------------------------------------------
OctoPrint version : 1.3.8
OctoPi version    : 0.14.0
------------------------------------------------------------------------------

pi@octopi-2:~ $ 
```
This is not what we want. 

If you look at the INVOCATION section of the **bash** manpage, there are different startup files sourced by the shell, depending on how they are invoked. (This is old and common shell behavior.)

Things that should be done on login belong in **.profile**. The "Welcome" message belongs there.

guysoft, 2018-04-17:
Again - If you have something that needs to be fixed please send a pull request. I can't follow guides, that's why I built CustomPiOS

closed 2019-04-23 by guysoft