guysoft/OctoPi #513: how to change the username "pi" to something else ?

amrithmmh, 2018-04-13:
i used `sudo passwd root` to chnage root password 
but when i login as root i get access denied 
how to change the username from pi?

John-Mc, 2018-04-13:
Not sure if this is an OctoPrint issue either. It might be an appropriate question for the [Raspberry Pi Forum](https://www.raspberrypi.org/forums/)

amrithmmh, 2018-04-13:
@John-Mc then why i cant change the root password?

John-Mc, 2018-04-13:
I don't know, since I never use the Root user or password. You do realize that user "pi" is not the same as the Root user, don't you? Most people log in as pi, and just use "sudo" when they need root-type access. To change the user pi password, just log in as pi, enter `passwd` on the command line and press enter, then follow the prompts.

Again, none of this is OctoPi or OctoPrint related. try a search or a request for help on the Raspberry Pi forum.

If you want to change your OctoPrint web access username and password, that's a different story and can be done through OctoPrint's web interface (log in to OctoPrint via a web browser as a user with admin access, click on the little black too icon hear the top center of the page, then go to access control in the left column. From there, you can add or delete users with a couple different levels of access.)

ChinookTx, 2018-04-13:
The root account is disabled by default on many distros.  I agree this is basic linux tasks and not at all related to Octo[pi-print]

amrithmmh, 2018-04-17:
@ChinookTx i want to know if by default it is disabled on Octopi so i asked here else i will ask in raspberry pi forum

midhunmonachan, 2021-07-27:
I changed the username, but I had to go to the octopi configuration to change the home directory url inorder for the octopi server to start

guysoft, 2021-07-27:
@midhunmonachan Why are you trying to change it in the first place?

midhunmonachan, 2021-07-28:
> @midhunmonachan Why are you trying to change it in the first place?

Well you are right. There is absolutely no need to do that. But I like to mess around and so do many others. I did this recently and succeeded. I just saw this page and thought I'll just let other's who are looking for the same results find what they are looking for. 

> i used `sudo passwd root` to change root password
> but when i login as root i get access denied
> how to change the username from pi?

This user couldn't login to root because of a config file in the raspberry pi which disables root login. Change it and voila.. it works. 
Next thing they would encounter is that when they go to the octoprint web page, it will say that the octopi server is down. There is an octopi server config file which points to pi's home directory. Replace pi with the new username and the server starts like normal.I haven't tested it after that but hope this helps someone