guysoft/OctoPi #514: Issue with octopi-network config on multiple Pis

boomshockpow, 2018-04-16:
I know is not a recommended board, but I tried to install octopi on the pi-zero W. and it was responsive, but had to be directly connected to the usb to SSH into it. Wifi was totally non-responsive. I attempted configuring Octopi-network as described in the setup page of octopi 0.14, and I have a WPA 2 protected connection. So I followed the instructions , put the SSID in the spot, etc. and it didn't work. So I actually changed my wifi settings temporarily to unprotected, and reflashed the pi with octopi 0.14 and did the unprotected wifi, with just the SSID. It didn't work. Finally I tried configuring the octopi-wpa_supplicant with both of these ways, and had no luck. So I thought "Hm, maybe this is just because this PI isn't recommended, that its wifi card is not supported. So I got a Pi B+ (Not a 2 or 3, but the original one) and dusted it off and threw the wifi dongle in it, and no luck with any of the options I described earlier. It only works over ethernet. It is not an issue with me finding it with bonjour, because octopi.local works fine over ethernet. It is not an issue with me not being on the same network, because my computer is connected to the same network.. Do you have any ideas what the problem could be? Should I manually edit the /etc/WPA_supplicant.conf? (If it's even in the same spot)

guysoft, 2018-04-16:
Not a bug, please use https://discourse.octoprint.org

closed 2018-04-16 by guysoft

John-Mc, 2018-04-16:
The discourse site has good thread on [getting WiFi connection working](https://discourse.octoprint.org/t/getting-your-wifi-connection-working/184)