guysoft/OctoPi #516: OctoPi 0.15.0 RC1 Status

guysoft, 2018-04-23:
Please try the release candidate.
Download it at:
https://storage.googleapis.com/octoprint/2018-04-23_2018-04-18-octopi-stretch-lite-0.15.0.zip

md5: ``e807a8899dc986114dede71b46d0d6ca``

Finally the planets have aligned, @foosel released OctoPrint 1.3.8, and Raspberrypi foundation released a pi and image that does not break everything (I hope).

### Changes in the image

* Raspberrypi 3B+ support
* `octopi-wpa-supplicant.txt` now default
* Bug fixes
* linux kernel 4.14 from new base image 2018-04-18-octopi-stretch-lite
* debian stretch from new base image

Developer Notes:
We have moved to build with [CustomPiOS](https://github.com/guysoft/CustomPiOS), which means that OctoPi has joined a larger build system which supports lots of modules and cool ways to configure and maintain debian-based IOT distos. If all is good here I hope also to release CustomPiOS v1.0.0

varazir, 2018-04-24:
Can your do a live upgrade?

guysoft, 2018-04-24:
No you can't.
However OctoPrint is upgradeable within your current installation.

alainm, 2018-04-24:
Octoprint 1.3.8 seems to have a memory leak problem, may I suggest the the final OctoPI release wait untill that is fixed?

John-Mc, 2018-04-24:
https://github.com/foosel/OctoPrint/issues/2585

foosel, 2018-04-24:
That memory leak that @John-Mc linked is not 1.3.8 but 1.3.9-dev aka `maintenance` branch (and also fixed). The hotfix release 1.3.8 is *not* affected by this and to my knowledge doesn't have any kind of leak problems.

John-Mc, 2018-04-24:
I guess I misread the original title of that thread.

FormerLurker, 2018-04-24:
I did a clean install on a 3B+ and so far so good!  It was so easy it practically installed itself.  I'll report back if I have any issues.

foosel, 2018-04-24:
@John-Mc nah, the ticket was worded confusingly (still contained a reference to "1.3.8-dev" even though that should now have been "1.3.9-dev"). I corrected that to avoid further confusion.

tdub415, 2018-04-24:
I just got a Pi3B+ the other day and got a stable nightly build up and running on it, I'll try out the release candidate on it when I get home tonight :)

rjlorimor, 2018-04-24:
I installed fresh with a 64G microSD on a Pi3B+ with no issues and all is running fine.

tdub415, 2018-04-25:
Seems to be running just fine for me on the Pi3B+, commented out Great Britain, uncommented US, added my wifi ssid, password, and uncommented the 4 lines regarding it, set my pi cam string.  Everything so far is working as expected.

tdub415, 2018-04-26:
Where you put your SSID and PSK in, did you also make sure to uncomment the line just above and below it?
Ex:
## WPA/WPA2 secured
network={
  ssid="NetworkName"
  psk="Password"
}

RiddicMEng, 2018-04-26:
Yes, of corse I did. Apparently the wifi module is fried on the board, so I sent it back, getting a new one next week.

tdub415, 2018-04-26:
Just making sure, I completely forgot to uncomment those on a nightly build a week ago and wanted to make sure that wasn't the problem :)

John-Mc, 2018-04-26:
I thought I remembered reading somewhere that if the WiFi country wasn't set properly on initial boot that the WiFi chip would be shut off, and that the state persisted on the next reboot. There were commands to reset it, but I don't recall them. If I recall correctly, this was a Raspian issue, not an OctoPi issue. I think Gina or Guy came up with some way to avoid the problem.

 I may be remembering this incorrectly, however.

MachWas, 2018-04-26:
Hello,
Raspberry is a new world for myself and I would like to use it to access my 3D Printer by Octopi.
Now  there have been countless attempts to set up the system thus make a new image on the sd card, while I made different images of them such as the Version that is based on Jessie, as one adapted for my printer model and the last one now is the Octopi Image based upon stretch.
My Router is a Centro Business 2.0 and I would like to use the the internal WiFi of the Pi. The power supply is the original Pi and the camera is connected to an active USB bus.

At all the installations I did try out, the WiFi of the Pi seemed to turn into Power mode and until the installation of the stretch based octopi version it has only be possible to get a power save mode off at "iwcondig" has been to do this manual over the console.

Now if I boot up my Raspberry and if I type in "iwconfig" the power save mode is listed as off. But nothingless the connection goes off, after a while and I'm real despaired......
I spend now more than 50 hours on it, I did read more than hundred postings about, took a look at I don't know anymore of the amount of videos but now I'm obviously lost.......

To solve this WiFi problem in a way, that beginners could solve it somehow would be a big wish from my part.

workshopninja, 2018-04-27:
Installed with no trouble on Pi3B+. Running my first print now - massively faster g-code upload than the zero-w I was running before.

cruzzer, 2018-04-28:
@guysoft  I've installed OctoPi 0.15.0 RC1 on my RPi 3B+. I'm seeing issues when streaming the Pi camera image over ssh (ssh_dispatch_run_fatal .... message authentication code incorrect ).

I'm a bit unsure where to place this, as there are many components involved (octoPi, raspian, pi drivers). What would you like me to do?

guysoft, 2018-04-28:
@cruzzer OctoPi does not come with ssh streaming. Its great that you built ot yourself, but I don't support it. You can ask in the community forum.

cruzzer, 2018-04-28:
Sure, thanks for the response.

chippypilot, 2018-04-28:
Tried it on a 3B and a 3B+. Uploaded and printed a file from each. No problems. 

Thanks

John-Mc, 2018-04-29:
@MachWas - what are you using for a power supply on your Pi?

MachWas, 2018-04-29:
@John-Mc 
Hi I'm using the original 2.5 Amp Pi Supply and for the cam an active USB hub.
AND at the command ifconfig the message says Power save mode = off, but nothing less the wifi of the board goes off. sometimes after a few minutes and sometimes after a few hours....

guysoft, 2018-04-30:
Released: https://github.com/guysoft/OctoPi/releases/tag/0.15.0

closed 2018-04-30 by guysoft