guysoft/OctoPi #524: Ideamaker Octoprint

ModMike, 2018-05-02:
I am trying to get Ideamaker to dump the file into Ocotprint Octopi version.  I tried the post processing script but it does not seem to work.  If I wanted to save it directly to my pi what path would I need?

John-Mc, 2018-05-03:
If I'm following your question correctly, you should set up a "watched folder" and dump your files into there. Here is a video on [setting up the Watched Folder](https://www.youtube.com/watch?v=RQ3ufPo95lg&t=0s&list=PLGc0ilxmLZs8UwoQMdpOkYF4gZMMu_Qro&index=3)

guysoft, 2018-05-03:
This is not a bug, please post in the community forum at http://discourse.octoprint.org/

@John-Mc thanks as always.

closed 2018-05-03 by guysoft

John-Mc, 2018-05-03:
Is there a way to make Github put up a dialog box every time someone starts a new issue? Something requiring them to confirm that they are making a bug report, not a help request. (Though I suspect that even that would not be 100% effective, as many new users are likely to consider any problem they experience a "bug")

guysoft, 2018-05-03:
Yes, but I can probably paste a like to the forum there