guysoft/OctoPi #526: Incompatibility between pip 10+ (as provided on OctoPi 0.15.0) and OctoPrint up to 1.3.8 causes issues

foosel, 2018-05-04:
#### What were you doing?

Installing a plugin from the plugin repository in OctoPrint 1.3.8 running on OctoPi 0.15.0. Steps to reproduce:

1. flash a fresh OctoPi 0.15.0 & boot it up
2. complete OctoPrint's first run wizard
3. install any plugin from OctoPrint's built in plugin manager (e.g. Custom Control Editor)
4. restart OctoPrint

#### What did you expect to happen?

OctoPrint reporting the plugin as installed. No warnings in the log on install & on restart.

#### What happened instead?

OctoPrint couldn't detect which plugin was installed and showed it as "Unknown" in the popup message. `octoprint.log` showed:

```
2018-05-03 05:16:29,062 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py:1643: UserWarning: No PKG-INFO found for package: octoprint_HeaterTimeout
  warnings.warn('No PKG-INFO found for package: %s' % self.package_name)
```

both after install and on startup during plugin detection. This is caused by a change in metadata of python packages installed through pip versions 10 and up, see also foosel/OctoPrint#2625. This problem is fixed on the `maintenance` branch.

Since OctoPi 0.15.0 however ships with OctoPrint 1.3.8 and pip 10 and is hence affected out of the box (and will stay affected on new installs even after OctoPrint 1.3.9 with the fix is released, until the user updates the included OctoPrint instance), I propose a release of OctoPi 0.15.1 with the `chroot_script` adjusted so that it upgrades the venv pip to 9.0.3 instead of the newest 10.0.1. I can provide a PR that does this, however I'm unsure against which branch since the change should naturally be done against 0.1.5.0, not current `devel` which might already contain more changes. So please tell me how to proceed here.

#### Did the same happen when running OctoPrint in safe mode?

Plugin installations and third party plugins as installed via pip are disabled when OctoPrint runs in safe mode, so this cannot be reproduced there, however it is a verified issues with pip 10+ installed plugins as seen in the above linked OctoPrint ticket.

#### Version of OctoPi

0.15.0

0.14.0 is still fine, as is an OctoPi 0.15.0 with pip downgraded to 9.0.3 via `~/oprint/bin/pip install pip==9.0.3`.

#### Printer model & used firmware incl. version

Issue isn't related to printer communication

#### Screenshot(s)/video(s) showing the problem:

```
2018-05-04 13:00:35,944 - py.warnings - WARNING - C:\Devel\OctoPrint\OctoPrint\src\octoprint\plugin\core.py:1643: UserWarning: No PKG-INFO found for package: octoprint_autoselect
  warnings.warn('No PKG-INFO found for package: %s' % self.package_name)

2018-05-04 13:00:35,969 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.
```

[image]

I have read the FAQ

guysoft, 2018-05-04:
0.15.1 it is then.
Will need to push a fix first

foosel, 2018-05-04:
So shall I just provide a PR against OctoPi's `devel` branch, or do we need a custom branch for that? Not sure about your hotfix process

foosel, 2018-05-04:
`devel` looks like it would be the wrong place since it already has the version incremented to `0.16.0`. Not sure if your Jenkins is prepared to build from another branch, if so I'd suggest creating a branch `hotfix` or something like that from the `0.15.0` tag and building the image from that.

guysoft, 2018-05-04:
There is a ``minor`` branch. You can use it

foosel, 2018-05-04:
The `minor` branch has diverged from `devel` (and is 4 years old). Objections to force pushing that to `0.15.0` first, then sending a PR against it to adjust the version to `0.15.1` and doing the needful with regards to `pip`?

guysoft, 2018-05-04:
Perhaps just
```
checkout 0.15.0
git branch minor-0.15.0
commit changes
```
I will release from there

foosel, 2018-05-04:
Done, see #527

omzart, 2018-05-05:
Hi, can you provide a way to solve a problem that is linked to this issue?
The problem is that the plugin installed with pip 10, can not be uninstalled.
"Uninstalling plugin "Bed Visualizer"
/home/pi/oprint/bin/python2 -m pip uninstall --yes octoprint_bedlevelvisualizer
Cannot uninstall requirement octoprint-bedlevelvisualizer, not installed
You are using pip version 9.0.3, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Done!"
It says that it is not installed, but it is still in my "Installed Plugins" list, while it is not, so I can not installed it again.

guysoft, 2018-05-06:
I have no way to test here, but you can try 
```
sudo pip install pip==9.0.3
```

foosel, 2018-05-06:
No! Not sudo! You need to change the pip of the venv! Connect via SSH, then

    ~/oprint/bin/pip install pip==9.0.3

guysoft, 2018-05-06:
What @foosel said ^

omzart, 2018-05-07:
Yes thanks you, it was already done, problem was that I could'nt install a plugin that was installed on pip 10 and uninstall on pip 9, because it was seen as "installed" but wasnt, I solved this by installing it manually ( from github page of the plugin )
Thanks anyway.

guysoft, 2018-05-09:
Please see and test https://github.com/guysoft/OctoPi/issues/530

closed 2018-05-29 by foosel

guysoft, 2019-01-27:
Ok, I am lifting this and preparing RC2

PR #527 (2018-05-04): Pin pip to 9.0.3 until OctoPrint 1.3.9 is released
Fixes #526
commit 26668e5a (2018-05-04): Pin pip to 9.0.3 until OctoPrint 1.3.9 is released

Fixes #526