guysoft/OctoPi #529: [Request] how about new streamer?

Vexatuspl, 2018-05-07:
Hi,

As you know a lot of people are having issues with making the mjpg-streamer work and as a matter of fact the project has been already abandoned.
Have you been thinking about moving to a different piece of software? I've heard some good opinions about motion for example.

Best regards
Greg

guysoft, 2018-05-07:
Do you have one in mind that is better?

Vexatuspl, 2018-05-07:
I've read of people using Motion with some success and it is at least an ongoing project:
https://github.com/Motion-Project/
I've also found this for of mjpg-streamer:
https://github.com/jacksonliam/mjpg-streamer
But as stated before - I'm seeing awfully lot of issues people are having with mjpg-streamer. Like not all the webcams are supported and even the supported ones do not always tend to work.

guysoft, 2018-05-07:
Motion is not a streamer, it takes photos only when the imagr has changed.
The second open you posted is what we use today: https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/config#L18

Again - is there aa different option at all?

hansaya, 2018-05-09:
I think UV4L is a good option. From what I heard UV4L better than mjpg-streamer but I personally haven't tried it yet. https://www.linux-projects.org/uv4l/

hansaya, 2018-05-09:
I ran into this interesting project as well: https://datarhei.github.io/restreamer/ https://www.raspberrypi.org/forums/viewtopic.php?t=132503

natiz, 2018-08-20:
I played around with UV4L a bit, seems pretty decent.
Haven't done much testing on performance or anything, but here is some info which might help push this thing forward. This is what I did to set this up:
```shell
curl http://www.linux-projects.org/listing/uv4l_repo/lrkey.asc | apt-key add -
echo "deb http://www.linux-projects.org/listing/uv4l_repo/raspbian/stretch stretch main" >> /etc/apt/sources.list
sudo apt-get update
sudo apt-get install uv4l uv4l-uvc v4l-decoder uv4l-encoder uv4l-renderer uv4l-server uv4l-webrtc
sudo uv4l -v -f --config-file=/etc/uv4l/uv4l-uvc.conf --driver uvc --device-id 046d:082d
```
This one if for USB camera, the setup of a picam is [a bit different](https://www.linux-projects.org/uv4l/installation/).
device-id should be taken from `lsusb`

The stream URLs are then available:
http://octopi.local:8090/stream/video.mjpeg
http://octopi.local:8090/stream/snapshot.jpeg?delay_s=0
The only issue I had is embedding it on octoprint - as it seems to add a timestamp to the url query param, which video.mjpeg then don't like. I then just commented it out from the dist js files (`packed_plugins.js` and `packed_core.js` located in `~/.octoprint/generated/webassets`)
The resolution can then be changed using the control panel: http://octopi.local:8090/panel

guysoft, 2019-08-20:
Gadget Angel posted on the community forum how to use ffmpegserver as a streamer, that is a good candidate:
https://community.octoprint.org/t/how-to-simultaneously-stream-3-streams-from-a-raspberry-pi-camera/11330?u=guysoft

I am kinda busy this days, so if anyone wants to write a script/test it more. That will speed stuff up.

guysoft, 2019-09-08:
Well I guess for not its not ffmpeg:
https://community.octoprint.org/t/how-to-simultaneously-stream-3-streams-from-a-raspberry-pi-camera/11330/17?u=guysoft

tl;dr 
Detailed tests show its not better.

hevilp, 2019-11-26:
> 
> 
> I played around with UV4L a bit, seems pretty decent.
> Haven't done much testing on performance or anything, but here is some info which might help push this thing forward. This is what I did to set this up:
> 
> ```shell
> curl http://www.linux-projects.org/listing/uv4l_repo/lrkey.asc | apt-key add -
> echo "deb http://www.linux-projects.org/listing/uv4l_repo/raspbian/stretch stretch main" >> /etc/apt/sources.list
> sudo apt-get update
> sudo apt-get install uv4l uv4l-uvc v4l-decoder uv4l-encoder uv4l-renderer uv4l-server uv4l-webrtc
> sudo service uv4l_uvc add 046d:082d
> ```
> 
> This one if for USB camera, the setup of a picam is [a bit different](https://www.linux-projects.org/uv4l/installation/).
> `046d:082d` being the cam device-id, which can be taken from `lsusb`
> 
> The stream URLs are then available:
> http://octopi.local:8090/stream/video.mjpeg
> http://octopi.local:8090/stream/snapshot.jpeg?delay_s=0
> The only issue I had is embedding it on octoprint - as it seems to add a timestamp to the url query param, which video.mjpeg then don't like. I then just commented it out from the dist js files (`packed_plugins.js` and `packed_core.js` located in `~/.octoprint/generated/webassets`)
> The resolution can then be changed using the control panel: http://octopi.local:8090/panel

Hi,

can you make plugin for octoprint? I'm not familiar to edit those files.

thanks in advance