guysoft/OctoPi #530: OctoPi 0.15.1 RC1 Status

guysoft, 2018-05-09:
Hey,
This is a hotfix pnning pip to version 9 because version 10 does not play nice with OctoPi plugins.

https://storage.googleapis.com/octoprint/2018-05-08_2018-04-18-octopi-stretch-lite-0.15.1.zip
Md5: ``4600d24c0618376c2a10c1507b8ba990``

mod38, 2018-05-10:
Installed RC1 on Pi 3 driving M3D Micro using the M33 Fio plugin. Got error message while connecting to the printer but server actually connected and a small test print was completed. 
The attached file is the last part of the log file created as the print finished. The section with the errors is a repeat of the same message which appeared several times in the log.

[OctoPi 0.15RC1 log segment.txt](https://github.com/guysoft/OctoPi/files/1993353/OctoPi.0.15RC1.log.segment.txt)

guysoft, 2018-05-11:
Says ``Bad file descriptor``. Cc @foosel.

workshopninja, 2018-05-11:
Installed on Pi3B Plus - have not tried a print yet but the "enclosure plugin" cannot read BME280 temperature sensor 
Getting Warning in log file (attached)
Also opened this in the enclosure plugin: More info on there
https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/139

[octoprint.log](https://github.com/guysoft/OctoPi/files/1996359/octoprint.log)

thanks

b-morgan, 2018-05-11:
I think it would be nice if there was a version.txt in the /boot directory that had the version number in it (0.15.1). A microSD card doesn't lend itself to writing a label on it and I've got two and can't tell which is which without booting them first.

guysoft, 2018-05-12:
Thrre is a version file in `` /etc/octopi_version``. Similar to Raapbian's ``/etc/issue``.

JenusL, 2018-05-13:
@guysoft Raspbian has a issue.txt in /BOOT. Would probably be good to have octopi_verion.txt there as well then.

guysoft, 2018-05-13:
@JenusL Point taken, cc @foosel 
Format is:
```
Raspberry Pi reference 2017-09-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 496e41575eeb9fa13f394ffb407b7bc1d00b21c2, stage2
```
We could release an RC2 with this. 
Have it as part of CustomPi OS 1.0.1

mod38, 2018-05-13:
My issue with an error message when connecting to the printer did not occur with OctoPi 0.14 and Octoprint 1.3.8. There was no change to the plugin so the only change was 0.15.1 RC1.

Thisismydigitalself, 2018-05-14:
I would test it but first need to know what's new from 0.15.0/1.3.8. 
I have installed a fresh SD with 0.15/1.3.8 and found it to be totally unusable. all my prints failed due to communication errors. If i only knew this 0.15.1 RC1 is aimed at solving (also) this issue, i would give it a try. can't say I was happy with 1.3.8 - it ruined 3 of my prints. I have reverted back to 1.3.6 (on this same 0.15 image) and everything is perfect - no more mid print halts and pauses (Smoothie based printer).

Await detailed description of what's new in 0.15.1.rc1. 
P.S. I am not a programmer.

guysoft, 2018-05-14:
@Thisismydigitalself read there chabgelog. There is only one change.
Your printer issues are OctoPrint, not OctoPi

guysoft, 2018-05-26:
I have no time, I'll be releasing this for now and then fix stuff like issue.txt which no one mentioned until this RC

guysoft, 2018-05-26:
Released:
https://github.com/guysoft/OctoPi/releases/tag/0.15.1

closed 2018-05-26 by guysoft