guysoft/OctoPi #531: Fresh install of 0.15 update/upgrade/samba no go

ZeekWorld, 2018-05-10:
I did a fresh installed on the latest 0.15 octopi.
I used to be able to run apt-get update successfully on prior builds but not with 0.15.
I've also configured samba to access the Upload folder of octoprint and it isn't working either.
Like I said above, all this was working on my previous version of Octopi.

pi@octopi:~ $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:2 http://archive.raspberrypi.org/debian stretch InRelease
Err:3 http://archive.raspberrypi.org/debian stretch Release
  503  Service Unavailable [IP: 93.93.135.188 80]
Reading package lists... Done
E: The repository 'http://archive.raspberrypi.org/debian stretch Release' does no longer have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
pi@octopi:~ $
pi@octopi:~ $

guysoft, 2018-05-11:
Not a bug,``503 Service Unavailable`` means your internet connection isn't working. Also, suggest you use 0.15.1 RC1 which has a hotfix

closed 2018-05-11 by guysoft

ZeekWorld, 2018-05-11:
No, the RPi is connected to the internet via wi-fi just fine.

pi@octopi:~ $ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=54 time=25.1 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=54 time=16.6 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=54 time=21.6 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=54 time=21.5 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=54 time=16.4 ms
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4006ms
rtt min/avg/max/mdev = 16.429/20.252/25.107/3.318 ms
pi@octopi:~ $ ping google.com
PING google.com (172.217.1.206) 56(84) bytes of data.
64 bytes from iad23s26-in-f206.1e100.net (172.217.1.206): icmp_seq=1 ttl=54 time                                                                                                                                                             =15.8 ms
64 bytes from iad23s26-in-f206.1e100.net (172.217.1.206): icmp_seq=2 ttl=54 time                                                                                                                                                             =17.8 ms
64 bytes from iad23s26-in-f206.1e100.net (172.217.1.206): icmp_seq=3 ttl=54 time                                                                                                                                                             =16.6 ms
64 bytes from iad23s26-in-f206.1e100.net (172.217.1.206): icmp_seq=4 ttl=54 time                                                                                                                                                             =17.3 ms

--- google.com ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3004ms
rtt min/avg/max/mdev = 15.844/16.937/17.891/0.788 ms
pi@octopi:~ $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:2 http://archive.raspberrypi.org/debian stretch InRelease
Err:3 http://archive.raspberrypi.org/debian stretch Release
  503  Service Unavailable [IP: 93.93.130.214 80]
Reading package lists... Done
E: The repository 'http://archive.raspberrypi.org/debian stretch Release' does n                                                                                                                                                             o longer have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disa                                                                                                                                                             bled by default.
N: See apt-secure(8) manpage for repository creation and user configuration deta                                                                                                                                                             ils.
pi@octopi:~ $

guysoft, 2018-05-13:
Can you try:
```
wget http://archive.raspberrypi.org/debian/dists/stretch/Release
```
 This is the fails it claims it can't download.

John-Mc, 2018-05-13:
I've had a similar thing happen when I had a weak WiFi signal. If I recall, I've also had the problem when my internet slows to a crawl even when my WiFi signal is strong (I've had some issues with my DSL feed, I also wonder if problems can be caused when my kids are hogging all the bandwidth on streaming multiple video feeds - my son's Xbox usage always seems to cause everyone else problems). Sometimes it would take 3 or 4 attempts of apt-get update/upgrade before it would download successfully. Moving the Pi to a location with a stronger WiFi signal helped some of these issues.

For the WiFi end of things, try the command `sudo iwlist wlan0 scan`. The beginning of the output should be something like this:
```
pi@octopi:~ $ sudo iwlist wlan0 scan
wlan0     Scan completed :
          Cell 01 - Address: F8:8E:85:C0:71:07
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=64/70  Signal level=-46 dBm  
                    Encryption key:on
                    ESSID:"YourWiFi SSID"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
```

Note that the above will show a separate block for every WiFi network that your Pi can see, whether it is connected to it or not. If there is more than one, just look for the one with the correct SSID. While you are looking, you can check for other networks on same channel/frequency: Multiple access points on the same channel can cause problems.

Pay attention to the "Quality" number: higher is better, the closer to 70 you get the better. Also look at the "Signal level": this is a negative number, and the more negative it goes, the weaker the signal. The numbers shown in the example above `Quality=64/70  Signal level=-46 dBm` give me good results. My old location for the Pi gave `Quality=29/70  Signal level=-81 dBm`. That caused me some problems.