guysoft/OctoPi #532: OctoPi Serial Port Failure During Print

JabawokJayUK, 2018-05-20:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2018-05-21:
Does it happen in OctoPi 0.15.x?

JabawokJayUK, 2018-05-21:
Issue observed in OctoPi 0.14.0

cassianoleal, 2018-05-22:
I get the same with OctoPi 0.15.0 and Octoprint 1.3.8. I've been running Octoprint 1.3.7 for a few hours now with no issues, but will report back if I have any.

taw-pi, 2018-06-26:
I am running OctoPi 0.15.1 and it happened to me :(

guysoft, 2018-06-26:
At this point its Beyond OctoPi. If there is any information you can contribute to https://github.com/foosel/OctoPrint/issues/2647 that might help @foosel figure out what the problem is.

taw-pi, 2018-06-27:
I did not have anything useful from the logs so I have enabled debug logging and am running the print again. I will add the logs if it fails again.

taw-pi, 2018-06-28:
Fortunately or unfortunately my print completed without error so I am not in a position to provide any further evidence of this fault. In the meantime I will leave debug options enabled and report if it happens again.

oliclarke, 2018-07-05:
Im not sure if Im having the same issue as in this thread. Printer randomly stops, fans + heat remain on, no error warning on marlin LCD screen.  Log here - > https://pastebin.com/HQYe3hdZ 

Running Marlin bugfix  1.1.9 
octoprint - Version 1.3.8

DerPauker, 2020-04-15:
I just had a very similar issue in version 1.4.0.

Here is the output of dmesg:
[432567.960338] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[432567.960430] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[432567.977156] usb 1-1.1.2: USB disconnect, device number 5
[432567.977508] cp210x ttyUSB0: failed set request 0x7 status: -19
[432567.977529] cp210x ttyUSB0: failed set request 0x12 status: -19
[432567.977541] cp210x ttyUSB0: failed set request 0x0 status: -19
[432567.977959] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[432567.978035] cp210x 1-1.1.2:1.0: device disconnected
[432818.384066] usb 1-1.1.2: new full-speed USB device number 6 using dwc_otg
[432818.521840] usb 1-1.1.2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[432818.521857] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[432818.521867] usb 1-1.1.2: Product: CP2102 USB to UART Bridge Controller
[432818.521876] usb 1-1.1.2: Manufacturer: Silicon Labs
[432818.521885] usb 1-1.1.2: SerialNumber: 0001
[432818.524887] cp210x 1-1.1.2:1.0: cp210x converter detected
[432818.531819] usb 1-1.1.2: cp210x converter now attached to ttyUSB0

The printer stopped printing, temperatures remained high and fans went to full speed.

I'm happy to provide more logfiles if required.

Regards
Manuel