guysoft/OctoPi #534: Remote Desktop (xrdp) works on 0.14.0, fails on 0.15.1

b-morgan, 2018-05-24:
#### What were you doing?

1. Install a fresh OctoPi 0.15.1 image on micro SD card
1. Edit octopi-wpa-supplicant.txt, etc.
1. Boot Raspberry Pi 3B+ from micro SD card
1. SSH into OctoPi
1. sudo apt-get update
1. sudo apt-get upgrade
1. ~/scripts/install-desktop
1. sudo apt-get install xrdp
1. On a Windows desktop, open Remote Desktop
1. Enter Octopi or IP address and Connect
1. Enter pi and password

#### What did you expect to happen?
A desktop display to appear in a window on the Windows desktop.

If you change the first step above to OctoPi 0.14.0, then that is exactly what happens.

If you change the first step above to Raspbian Stretch, then it also works. I have not tried Raspbian Stretch Lite (not sure if there's a desktop installed and if not, how to install it).

#### What happened instead?
I get a blank screen. Sometimes if I wait long enough I'll get this as well:

[image]

#### Did the same happen when running OctoPrint in safe mode?
N/A. This is strictly an OS issue

#### Version of OctoPi
0.15.1

#### Printer model & used firmware incl. version
N/A

#### Screenshot(s)/video(s) showing the problem:
Probably not needed... its just a window with a solid background color.

Yes, I have read the FAQ.

guysoft, 2018-05-26:
The screen says that you are trying to connect to localhost ``127.0.0.1``. Not your Pi.
Please use the community forum, because this is not part of OctoPi's functionality: http://discourse.octoprint.org/

b-morgan, 2018-05-26:
I don't quite understand the exact process of connecting to the Pi with Remote Desktop but I believe various processes are passed around internally with the end result being a display of the Pi desktop on the remote system's screen. Therefore, **I** am not trying to connect to localhost, some part of the process is.

Please try and reproduce this for yourself. I believe your skills at figuring out what is going wrong are far superior to mine. If you think it would be worthwhile for me to reproduce using Raspbian Stretch Lite, let me know (and given your superior knowledge of Stretch Lite, hints on what prerequisites will be needed would be very helpful). If there are other log files that might shed more light on the problem, tell me what files you need and I'll provide them.

guysoft, 2018-05-26:
I am not going to reproduce this, I am not working for you, please refer to the community forum.

b-morgan, 2018-05-26:
Sorry to have offended you as it was not my intent. 

I did attempt to get help from the [community forum](https://discourse.octoprint.org/t/octopi-0-15-1-install-desktop-script-xrdp-on-rpi-3b/1906) prior to coming here but the expertise of those that responded were not able to arrive at a solution. I came here because it is more of an OS issue than an OctoPrint issue and was hoping your expertise would be able to provide a solution or point me to a better place to find one.

John-Mc, 2018-05-27:
What Guy is saying is that this is not an OctoPi issue. He does not have the time to help people troubleshoot unrelated issues. In fact, this forum is not really a help/troubleshooting forum, it is for reporting bugs in the OctoPi image.

The discourse forum he linked is a good place to start for help with things related to OctoPrint/OctoPi. If they were unable to help you there, and if as you seemed to indicate it may be an issue with Stretch Lite, you might try the [Raspberry Pi Forums](https://www.raspberrypi.org/forums/). If you scroll down that page, you'll find a subforum dealing with Raspian (though there may also be other subforums related to your issue.)

closed 2018-06-09 by guysoft