guysoft/OctoPi #535: Octopi and 5G

bill-orange, 2018-05-25:
#### What were you doing? Trying to connect to my 5G network. I can connect but when I edit the config file to include the -5G suffix, no connection.  This is using a B+ PI

1. Using a network that includes a 5G connection and using a PI B+, first connect not using 5G and confirm success.
2. Edit config to the 5G SSID and try to connect
3. Failure!

#### What did you expect to happen?  5G connection as in RASPIAN

#### What happened instead? 2.4G connection okay and no 5G

#### Did the same happen when running OctoPrint in safe mode?  Did not test, Not relevant

#### Version of OctoPi Latest RC candidate. 1.3.8/0.150

P.S. I also retested after:

apt-get update
apt-get upgrade
rpi-update

guysoft, 2018-05-25:
Dod you set your country codr correctly? Other than that I have no way of testing.  I don't have 5G here.

Penfold81, 2018-05-25:
Do you mean 5G wireless band from your router? I have mine set on 2.4G as it’s all that the pi3 would pick up

bill-orange, 2018-05-25:
Yes, the 5G wireless band from the router. With Raspian, my B+ picks it up just fine. 

Guysoft, I don’t see a country code setting in the supplicant config. DidnInmiss it?

guysoft, 2018-05-25:
Its at the bottom

bill-orange, 2018-05-25:
Missed it then. I will set it and report back.

bill-orange, 2018-05-25:
@guysoft Good catch! It was the country code.  I am in US default was GB.

closed 2018-05-25 by bill-orange