guysoft/OctoPi #537: [ERROR] OctoPi broken DL button?

bolsoncerrado, 2018-05-30:
Tried to download OctoPi 0.15 today, got this instead:

<Error>
<Code>AccessDenied</Code>
<Message>Access denied.</Message>
<Details>
Anonymous caller does not have storage.objects.get access to octoprint/raspberry/stable/octopi-stretch-lite-0.15.1.zip.
</Details>
</Error>

guysoft, 2018-05-30:
Can yoi please link ti the download button you are using?  It works for me on my phone just now.

bolsoncerrado, 2018-05-30:
The one on the OctoPi page. Let me try again.

bolsoncerrado, 2018-05-30:
Just tried again, same error. Tries to access:

https://storage.googleapis.com/octoprint/raspberry/stable/octopi-stretch-lite-0.15.1.zip

bolsoncerrado, 2018-05-30:
Perhaps as the owner you're able to access it? The error mentions "Anonymous caller":

Anonymous caller does not have storage.objects.get access to octoprint/raspberry/stable/octopi-stretch-lite-0.15.1.zip.

bolsoncerrado, 2018-05-30:
This XML file does not appear to have any style information associated with it. The document tree is shown below.

John-Mc, 2018-05-30:
I just tried it from the Download button on this page: https://github.com/guysoft/OctoPi
It worked fine for me.
I get the same error when using the link that @bolsoncerrado used.

bolsoncerrado, 2018-05-30:
The error comes from this button on this page:

https://octoprint.org/download/

bolsoncerrado, 2018-05-30:
...and I just retried and it works now :?

John-Mc, 2018-05-30:
That link works for me, however, when I click on the link in your post, I get the same error you did.

guysoft, 2018-05-30:
Thia works: https://storage.googleapis.com/octoprint/stable/octopi-stretch-lite-0.15.1.zip

Cc @foosel the /stable is the one displayed in the nightly pages. While raspberypi/stable is what on the download link

foosel, 2018-05-30:
@guysoft weird. The prior one was at `raspberry/stable`. Any changes in your upload scripts?

In any case, should be fixed in the load balancing config:

```
$ curl -I https://octopi.octoprint.org/latest
HTTP/1.1 302 Moved Temporarily
Server: nginx
Date: Wed, 30 May 2018 15:03:15 GMT
Content-Type: text/html
Content-Length: 154
Connection: keep-alive
Location: https://storage.googleapis.com/octoprint/stable/octopi-stretch-lite-0.15.1.zip

$ curl -I https://storage.googleapis.com/octoprint/stable/octopi-stretch-lite-0.15.1.zip
HTTP/2 200
expires: Wed, 30 May 2018 16:03:30 GMT
date: Wed, 30 May 2018 15:03:30 GMT
cache-control: public, max-age=3600
last-modified: Sat, 26 May 2018 11:03:40 GMT
etag: "4600d24c0618376c2a10c1507b8ba990"
[...]
content-type: application/zip
content-disposition: attachment;filename="octopi-stretch-lite-0.15.1.zip";filename*=UTF-8''octopi-stretch-lite-0.15.1.zip
[...]
```

closed 2018-06-09 by guysoft