guysoft/OctoPi #548: [Request] Include Octoprint-TFT as optional configuration

mcuadros, 2018-09-12:
I am the author of Octoprint-TFT a OctoPrint touch interface for TFT touch modules based on GTK+3, https://github.com/mcuadros/OctoPrint-TFT

Still is not working perfect, since has some bugs, but many people find this project very useful to control the Ocotoprint server.

The main issue with my users is that even I provide dpkg with the configuration, is not that obvious for some users, as well as I am not an expert building X services for Debian, and the package is not working perfectly.

I really think that could help a lot to the users having Octoprint-TFT as a package shipped with OctoPi. 

If this is something acceptable, I can work in a PR to OctoPi, including Octoprint-TFT

guysoft, 2018-09-12:
We shipped at the time [OctoPiPanel](https://github.com/jonaslorander/OctoPiPanel), which is unmiantained anymore and got removed.
I will accept a PR, but only if it comes with a willingness to maintain it and keep it working.
We might also want to ship an installer script, so X services would not be bundled to the image and bloat it.

guysoft, 2019-07-02:
Is this still being maintained?
Closing if no reply.