guysoft/OctoPi #549: Restart/reboot commands are not working

adrianmihalko, 2018-09-12:
#### What were you doing?

On a fresh installation on Raspberry Pi, built in restart commands (Reboot system, Restart OctoPrint) are not working.

#### What did you expect to happen?

I try to restart/reboot server/OctoPrint from the top menu.

#### What happened instead?

Nothing.

#### Did the same happen when running OctoPrint in safe mode?

Nothing.

#### Version of OctoPi

OctoPrint version : 1.3.8
OctoPi version    : 0.15.1

#### Printer model & used firmware incl. version

LOG:

Reboot command:

```
2018-09-12 18:25:53,327 - octoprint.server.api.system - INFO - Performing command for core:reboot: reboot
2018-09-12 18:25:53,460 - octoprint.server.api.system - WARNING - Command for core:reboot failed with return code 1:
! STDOUT: 
! STDERR: Failed to set wall message, ignoring: Interactive authentication required.
! Failed to reboot system via logind: Interactive authentication required.
! Failed to open /dev/initctl: Permission denied
! Failed to talk to init daemon.
```

Restart Octoprint:

```
2018-09-12 18:28:25,535 - octoprint.server.api.system - INFO - Performing command for core:restart: service octoprint restart
2018-09-12 18:28:25,679 - octoprint.server.api.system - WARNING - Command for core:restart failed with return code 1:
! STDOUT: 
! STDERR: Failed to restart octoprint.service: Interactive authentication required.
! See system logs and 'systemctl status octoprint.service' for details.
```

So it's definitely an authentication problem, but how do I resolve this?

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2018-09-12:
Cc @foosel

adrianmihalko, 2018-09-19:
@guysoft @foosel Should I open a ticket on OctoPrint's GitHub or this problem is related to OctoPi?

foosel, 2018-09-19:
That's on OctoPi. The problem is that OctoPrint can't execute the command due to this "Interactive authentication required" error thrown by the restart command. OctoPrint is executing the command just fine, but the underlying os refuses to restart.

Never seen this error though, and fairly sure that I've restarted on 0.15.1 at least 50 times by now during update tests, so no idea right now.

Could be something that broke due to a dist upgrade maybe in which case it smells like a breaking change in raspbian. But hard to say.

adrianmihalko, 2018-09-19:
I don't know, I am on pure/fresh install of OctoPi, so the bug is not on the user side. I think.

adrianmihalko, 2018-09-19:
I think the problem is with the sudoers file.

```
pi@octopi:~ $ ls /etc/sudoers.d/
010_pi-nopasswd  octoprint-service  octoprint-shutdown  README
pi@octopi:~ $ cat /etc/sudoers.d/octoprint-service 
pi ALL=NOPASSWD: /usr/sbin/service
pi@octopi:~ $ cat /etc/sudoers.d/octoprint-shutdown 
pi ALL=NOPASSWD: /sbin/shutdown
pi@octopi:~ $ cat /etc/sudoers.d/010_pi-nopasswd 
cat: /etc/sudoers.d/010_pi-nopasswd: Permission denied
pi@octopi:~ $ sudo !!
sudo cat /etc/sudoers.d/010_pi-nopasswd 
#pi ALL=(ALL) NOPASSWD: ALL
```

This last file should be uncommented?

foosel, 2018-09-20:
That last file would allow *every* command to be executed as user root without a password. It's intentionally commented since that is dangerous in the wrong hands (which is why it makes me shake my head that it's the default in Raspbian).

The relevant ones here are `octoprint-service` (passwordless `sudo service octoprint restart`) and `octoprint-shutdown` (passwordless `sudo shutdown -h now` and `sudo shutdown -r now`), which should be the commands it tried to execute here when it failed.

What does happen when you SSH in as user `pi` and run `sudo service octoprint restart`, `sudo shutdown -h now` and `sudo shutdown -r now` (note that the last two will cause the pi to shutdown/reboot)? Do you also get a permission error then?

For reference:

```
------------------------------------------------------------------------------
OctoPrint version : 1.3.9
OctoPi version    : 0.15.1
------------------------------------------------------------------------------

pi@octopi2:~ $ sudo service octoprint restart
pi@octopi2:~ $ cat /etc/sudoers.d/octoprint-*
pi ALL=NOPASSWD: /usr/sbin/service
pi ALL=NOPASSWD: /sbin/shutdown
pi@octopi2:~ $
```

Also works fine from the web interface.

adrianmihalko, 2018-09-20:
Of course If I manually execute these commands with `sudo`, all they are working fine.

pi@octopi:~/.octoprint/logs $ sudo service octoprint restart
pi@octopi:~/.octoprint/logs $ sudo cat /etc/sudoers.d/octoprint-*
pi ALL=NOPASSWD: /usr/sbin/service
pi ALL=NOPASSWD: /sbin/shutdown

ChinookTx, 2018-09-20:
Have you tried logging out and logging back in to the web interface?

adrianmihalko, 2018-09-20:
Yes, but this is not related. If things would work correctly I would able to restart service without `sudo` from command line:

```
pi@octopi:~/.octoprint/logs $ service octoprint restart
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart 'octoprint.service'.
Authenticating as: root
Password: 
polkit-agent-helper-1: pam_authenticate failed: Authentication failure
==== AUTHENTICATION FAILED ===
Failed to restart octoprint.service: Access denied
See system logs and 'systemctl status octoprint.service' for details.
```

I am actively searching why /etc/sudoers.d/* content is not used by current OctoPi.

adrianmihalko, 2018-09-20:
I found the issue:

[image]

`sudo` command is missing before these commands. I don't remember If I removed them or it is like this by default, but this was the problem.

foosel, 2018-09-21:
> I don't remember If I removed them or it was like this by default

My money would be on you modified those lines because [this is definitely how they ship](https://github.com/guysoft/OctoPi/blob/0.15.1/src/modules/octopi/filesystem/home/pi/.octoprint/config.yaml#L12-L14). So not that fresh of an install after all ;)

closed 2018-09-21 by adrianmihalko