guysoft/OctoPi #550: webcam device validation too strict

ej0rge, 2018-09-21:
#### What were you doing?

using udev rules to make my webcams predictable

#### What did you expect to happen?

well i didn't expect it to be this hard

#### What happened instead?

any usb_device_path not matching /dev/video[0-9]\+\ is replaced by /dev/video0

#### Did the same happen when running OctoPrint in safe mode?

N/A

#### Version of OctoPi

0.15.1

#### Printer model & used firmware incl. version

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

In order to avoid namespace conflicts i am under the impression that it is normal to use nonstandard device names when using udev rules to create symlinks pointing to devices with specific characteristics, so as to avoid random renumbering of devices if they are plugged into a different usb port or just happen to identify in a different order upon reboot. 

Had to change the regex to /dev/video\w\+\ in order for the descriptive symlinks created by my udev rules to be honored. e.g. /dev/videoX5S for the camera attached to the X5S.

guysoft, 2018-09-21:
Do you have a PR to suggest a solution that would work for others?
Because I don't know what would solve a scenario using non-standard naming conventions (which is set by Debian).

Why not just use /dev/video9?

ej0rge, 2018-09-22:
To be honest I am not sure what dread condition you are hoping to prevent by validating the usb device path. 

I was setting up two instances of octoprint on my pi3 last night and fought with this for an hour before figuring out why only /dev/video0 would get streamed no matter which instances of webcamd was fired up. 

The existing guides on the internet are all pre-02/18 and say to edit the options line which of course has no effect. 

usb_device_path doesn't seem to be mentioned in any documentation, but then setting it to the symlink i created with a udev rule to make sure it can find the right camera every time was ignored. 

why not use /dev/video9? Why use it?

guysoft, 2018-09-23:
I suggested using `` /dev/video9`` because it should be captured by the prefix, and because I was assuming you are using less than 10 cameras.
Again, I don't seem to understand what would solve the issue for you, if you have any suggestion you are welcome to provide a pull request.