guysoft/OctoPi #552: ssh: connect to host ... port 22: Connection refused

atar-axis, 2018-10-06:
#### What were you doing?

1. Downloaded latest octopi-image
2. uploaded via `sudo dd bs=512k if=./2018-04-18-octopi-stretch-lite-0.15.1.img  of=/dev/mmcblk1 status=progress && sync`
3. inserted into raspberry pi 3b+ and plugged lan cable in
4. tried to connect via `ssh pi@ip-address-as-shown-on-my-routers-webinterface`

#### What did you expect to happen?

A working connection

#### What happened instead?

`ssh: connect to host ip-address port 22: Connection refused`

#### Version of OctoPi

15.1

Penfold81, 2018-10-07:
Port 22? Have you tried port 80?

Thanks

Dave

guysoft, 2018-10-07:
Do you get the same issue when you use Raspbian lite?

atar-axis, 2018-10-07:
You mean a clean raspbian image? 
Unfortunately not, everthing works fine then.

guysoft, 2018-10-07:
I don't understand? You did try and it work? or you didn't?

At the moment you are saying that something that worked for thousands of people, does not work for you. So I have no way to pinpoint there is an error.

atar-axis, 2018-10-07:
Sorry, your question was if I get the same error using raspbian lite, and my answer was "unfortunately not", so yes - i tried it (a clean raspbian image) and it worked just as expected.
But the octopi image does unfortunately not.

atar-axis, 2018-10-07:
I found the problem...
I checked the `md5sum` of the .zip file, and it was correct - but i never checked the md5 of the .img - since it was not published on the website. Unfortunately something went wrong while extracting...

So for everybody else, the `md5sum` is `2f7cd2a9aa46464595c0983fe0563a34` for the `2018-04-18-octopi-stretch-lite-0.15.1.img` file.

closed 2018-10-07 by atar-axis