guysoft/OctoPi #555: The "pi" user does not exist on armbian by default.

mr-miky, 2018-10-28:
In raspbian the image of the uSD already contains the default  user **pi** with **raspberry** password and is among the suders.

On armbian the image of the uSD does not have users except root and at the first login it allows to create the "normal" user. After connecting in ssh, create the user pi with the raspberry password and insert it into the sudoers.

The problem is that the custompios script starts from the assumption that pi exists and changes the group membership, the mode of operation of the sudo etc causing the failure of the script since it does not find p among the sudoers ... .

In practice you need to write "armbian" for OrangePiZero on a uSD, start OrangePiZero, create the user pi, get the image from the uSD and pass it to custompios as an image of the variant.

guysoft, 2018-10-30:
What are you trying to do?
Are you building or using an existing build?
Of building,  please attach ```build.log``

guysoft, 2018-11-03:
Closing due to no response

closed 2019-04-23 by guysoft