guysoft/OctoPi #556: Octoprint 1.3.9 crash during print, python attribute error

trohnjavolta, 2018-11-10:
#### What were you doing?
Sending gcode from simplify3d to octopi running on raspberry pi zero 
[octoprint.log](https://github.com/guysoft/OctoPi/files/2568240/octoprint.log)
and starting the print.

#### What did you expect to happen?
Octopi finishing the print without problems or crashes.
#### What happened instead?
About 1,5 hours in print, octoprint crashed, didn't restart. After manual restart data is gone and I cannot resume the print.

#### Did the same happen when running OctoPrint in safe mode?
So far I did not test it in safe mode.

#### Version of OctoPi
OctoPrint 1.3.9 running on OctoPi 0.15.1

#### Printer model & used firmware incl. version
Creality Ender 3 with Marlin Firmware, TH3D flavour. Version TH3D_UFW_U1.R2.3

#### Screenshot(s)/video(s) showing the problem:
octoprint.log attached. Relevant lines: 144 - 164

guysoft, 2018-11-11:
This is an OctoPrint issue, not an OctoPi, issue, please post it to https://github.com/foosel/OctoPrint/issues/

closed 2018-11-11 by guysoft

trohnjavolta, 2018-11-11:
:D Ok...on OctoPrint github I read no issues should be posted there if using octopi. I'll post it there.