guysoft/OctoPi #558: first run - invalid credential

Gigio2k, 2018-11-21:
#### What were you doing?
1. download image of octopi from https://octoprint.org/download/
2. setup wifi 
3. build sdcard and boot up
4. access via ssh with username: pi passowrd: raspberry **[OK]**
**5. connect to web at : http/{ip}**
[redirect to https with cert warning to login.cgi]
**6. login as pi/raspberry** 
**7. get ERROR INVALID CREDENTIALS

**

#### Version of OctoPi
OctoPi 0.15.1 

[octopi.log](https://github.com/guysoft/OctoPi/files/2604749/octopi.log)

[image]

guysoft, 2018-11-21:
The page is not OctoPrint page, you are routed to a different machine. Its
an issue with your local network setup

Gigio2k, 2018-11-21:
really ? the router show me this ip, the connection with ssh work with this ip, there is some settings to change the ip and try ? 
(now i flash again the card and restart)

John-Mc, 2018-11-21:
@Gigio2k - That is not the page you would see if you were logging in to OctoPrint

guysoft, 2018-11-21:
An issue can't be deleted,  only closed