guysoft/OctoPi #559: RaspberryPi 3B+ w/2.1 Camera Not working

jhihn, 2018-11-24:
I downloaded the latest release of octopi (my first OctoPi experience), installed it on a Raspber Pi 3b+ (brand new) with a 2.1 camera module.

The camera was detected but would not work:

    pi@octopi:~ $ vcgencmd get_camera
    supported=1 detected=1
    pi@octopi:~ $ raspistill -o /tmp/foo.img
    mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
    mmal: camera component couldn't be enabled
    mmal: main: Failed to create camera component
    mmal: Failed to run camera app. Please check for firmware updates

I fixed it with:

    pi@octopi:~ $ sudo rpi-update
    pi@octopi:~ $ sudo reboot

guysoft, 2018-11-24:
You didn't provide the version of OctoPi. Have you tried the nightly?
I have no 2.1 camera module to test against.

guysoft, 2018-11-26:
The release announcement of the camera is after OctoPi 0.15.1 is released, although there are no release notes saying anything about the camera. I suggest you try the nightly and report. That should work. And we could add it to our release notes.

guysoft, 2019-07-02:
Closing if no reply

jhihn, 2019-07-02:
Worked eventually.

closed 2019-07-02 by jhihn