guysoft/OctoPi #560: Can't WiFi Turn off using desktop's WiFi icon at Raspberry Pi Jesse

joonghochoe, 2018-11-28:
#### What were you doing?

1. Using OctoPi + CustomPiOS + 2017-07-05-raspberry-jessie desktop version image, build OK.
2. Boot OctoPi image
3. Turn off WiFi with WiFi icon on the right side of the taskbar, But WiFi icon still remains On

#### What did you expect to happen?
I must use Jessie Desktop environment and use WiFi icon for WiFi On/Off.

#### What happened instead?
I can't turn off WiFi with WiFi icon of taskbar.

#### Did the same happen when running OctoPrint in safe mode?
It does not seem to have anything to do with OctoPrint safe mode.
1. Original Jessie version : can turn off WiFi
2. Original Strech version : can turn off WiFi
3. Strech + Octopi : can turn off WiFi
4. Jessie + Octopi : can't turn off WiFi

#### Version of OctoPi
0.15.1

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2018-11-28:
Is this an OctoPi build you built on your own? 
If so please provide build log.
Why are you using Jesse?

joonghochoe, 2018-11-28:
Yes, I'm making it for myself. I'm making UI with python+wxpython(old 2.9 version, not available at stretch). I tested Jessie + Octopi(with mjpg streamer, HAProxy, WiringPi or without), but result is all same.

[build.log](https://github.com/guysoft/OctoPi/files/2623423/build.log)

Do you have any idea?

guysoft, 2018-11-28:
Using jessie is insecure since it is not supposed anymore by debian. It would make far more sense for you to compile wxpython 2.9 against modern stretch than use legacy jessie.
I cannot support Jessie, its really old, even Debian don't support it.