guysoft/OctoPi #561: Raspberry Pi 3 Model A+ Octopi compatability

lrv209, 2018-11-29:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

lrv209, 2018-11-29:
Hello, I have seen that OctoPi is not yet compatible with the Raspberry Pi 3 Model A+ but that is what I have so am wondering if there is yet any indication on if/when the wonderful Octopi  install might become available for an installation on this Pi?

Hope someone can give me an indication so I can plan my more complicated install using octoprint.
Regards

guysoft, 2018-11-29:
Please use for now the nightly build at:
https://gnethomelinux.com/OctoPi/nightly/

A+ was released after the current release of OctoPi, and we wont wont make a new release until OctoPrint 1.3.10 rolls out (Currently RC3).

aaronvegh, 2018-12-29:
Alas, the server hosting the nightly builds is currently 503-ing.

guysoft, 2019-01-01:
@aaronvegh Sorry about that, I moved apartments and physically had to move the build server. Its now sitting in a temporally location until I move in to a new apartment, hopefully in two weeks.

This is also a good time to mention that if anyone wants to donate a build server in the cloud it would be great! The new docker build supports it.

omnislash772, 2019-01-12:
Looks like gnethomelinux goes to a music forum now

guysoft, 2019-01-12:
gnethomelinux.com goes to that forum for the past 15 years. There are
separate containers on that server, nightly is on a different path.

omnislash772, 2019-01-13:
Ah, I guess the container for it is down then because I get this.
`The requested URL /octopi/nightly/ was not found on this server.`

guysoft, 2019-01-13:
Works, for me, have a look at https://gnethomelinux.com/OctoPi/nightly/
Might have been a hickup

jasonhazel, 2019-01-22:
Looks like gnethomelinux.com might be down again.  I've been getting a 504 error from Nginx for the past day.

guysoft, 2019-01-22:
Indeed it is. I think the VPN got disconnected and I need to drive to my
parents place to fix it.

tedder, 2019-01-22:
Guy, I pinged you in discourse to at least have a backup of builds and hosting for you.

jasonhazel, 2019-01-22:
I've got some pre-paid credit at a VPS host (ChunkHost) that I'm not using.  It's enough to cover a 2GB Ram, 30GB Storage and 6TB Bandwidth for a year.  Would those resources suffice for another build server?

guysoft, 2019-01-22:
Yes, that should be more than enough

guysoft, 2019-01-22:
Ah wait, no. A year is nothing. I am not switching my current build system
for something that lasts only a year.

DanoDataScientist, 2019-01-24:
Guy, any word on when the nightly’s will be back up?

guysoft, 2019-01-24:
Fixed, its back up. I hope also to release next week an RC so the 3 A+ has a stable download. So hope these glitches would not be as urgent.

guysoft, 2019-01-24:
Ok, RC is out: 
https://github.com/guysoft/OctoPi/issues/571

wes1993, 2021-10-07:
Hello,
Is the latest release compatible with Raspberry PI Model A+?

guysoft, 2021-10-07:
This has been fixed in the release

closed 2021-10-07 by guysoft