guysoft/OctoPi #563: Upgrade use ssh from 1.3.9 to 1.3.10

userosos, 2018-12-11:
Hello! Everyone! I can not upgrade my octoprint from 1.3.9 to 1.3.10 from web. How can i get 1.3.10 use ssh? Sorry for noobs question.

guysoft, 2018-12-11:
1.3.10 is not released yet. For information on how to upgrade to RC see https://discourse.octoprint.org/t/how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image/206

closed 2018-12-11 by guysoft

John-Mc, 2018-12-12:
1.3.10 was released on December 10.
[](https://octoprint.org/blog/2018/12/10/new-release-1.3.10/)

guysoft, 2018-12-12:
Spoke to @foosel two days ago and she said "release today" and missed that.
Reguardless, this is an OctoPrint issue, not an OctoPi issue, so you should report it to https://github.com/foosel/OctoPrint/issues/

foosel, 2018-12-12:
[How to update manually via SSH is explained in the FAQ](https://discourse.octoprint.org/t/how-can-i-update-the-octoprint-installation-on-my-octopi-image/207).