guysoft/OctoPi #567: Canakit says P3B+ problem is OS issue

chilinski, 2019-01-15:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

chilinski, 2019-01-15:
I got an email today from Canakit because I bought a Rasp3b+ from Amazon. The email said users of 3B+ are getting the lightning bolt icon in Octoprint not because of an inadequate PSU but because the OS doesn't support the 3B+. I get the "undervolted" icon/message, and I ordered a beefier PSU to drive it (which I wanted any way). So, just reporting what Canakit is telling its 3B+ customers.

chilinski, 2019-01-15:
>    1. Do you actually see thr undervolt icon when you use their equipment?
>    2. What version of OctoPi?
>    3. Can you post the full content of the email?

Here is the email from Canakit:

Thank you for your purchase of the Raspberry Pi 3 Model B+ from CanaKit.

IMPORTANT NOTES:

1. If you see lightning bolt symbol with a rainbow splash screen upon
boot it indicates that you are using an operating system that is NOT
compatible with the "B+" model.  It is NOT related to the power supply
voltage.

2. The "B+" model requires "B+" compatible software to operate (such
as NOOBS 2.7.0 or later).  Please ensure you are using a compatible
operating system or the board will not boot.

3. You must ensure the MicroSD card you are using is formatted as
FAT32.  You can use the "SD Formatter" tool to format cards up to 32GB
in size.  When formatting a 64GB or larger card on a PC, you must use
the "FAT32 Format" tool to format the card.

Our goal at CanaKit is for you to be completely satisfied with both
the service you have received and the product you purchased.  Please
let us know if there is any way we can assist. If you have any issues,
simply reply to this email, and we will be happy to ensure your
complete satisfaction with CanaKit.

We hope you enjoy your purchase, and thanks again for choosing CanaKit!

guysoft, 2019-01-15:
Ok,
1. You wrote ``The email said users of 3B+ are getting the lightning bolt icon in Octoprint``. However they didn't there is no mentioning of OctoPrint or OctoPi int he email. Therefore we can see if the claim holds by comparing to Raspbian.
2. Can you boot an official Raspbian from Raspberrypi site and see if you get the lightning bolt symbol there? OctoPi is based on raspbian, so if it shows up there too, then this is not an OctoPi issue, and its an issue of your hardware supplier providing false information. Otherwise OctoPi might be another issue.

chilinski, 2019-01-15:
Follow up: You are correct that Canakit did NOT specifically mention
Octoprint. That was my bad wording.

I see the indicator in Octoprint. I see it also when I boot the 3B+ in an
attached console.

chilinski, 2019-01-15:
One last thing, when you click the undervolt icon in Octoprint, you get
this link from Gina:

https://discourse.octoprint.org/t/i-had-a-strange-lightning-bolt-temperature-symbol-with-an-exclamation-mark-popup-in-my-navbar-what-does-that-mean/5256

guysoft, 2019-01-15:
The undervolt icon is managed by OctoPrint and not by OctoPi. If you have an issue with it please report it there.

I suggest you only address one issue per issue, otherwise it will be a mess.

chilinski, 2019-01-15:
Fine. I'm done here.
I see the issue in Octoprint.
Octoprint ,via Gina, says it's a psu issue
Canakit says it's not a psu issue, it's an os issue.
Octopi provides the OS build.

John-Mc, 2019-01-15:
The current version of OctoPi ships with a Raspian Lite OS that **is** compatible with the Pi. If your Pi is booting up, you have the right OS. The Canakit explanation is probably referring to older versions of Raspian (whoch would not even let the Pi 3B+ boot.

guysoft, 2019-01-15:
I think @John-Mc intepretion is right, its on the same bullet referring to the rainbow. Unless there is more helpful input I am closing this.

John-Mc, 2019-01-15:
I bought 3 Pis from Canakit which came in a kit that included the power supply. Two of them worked fine, one of them would not provide full voltage under load. I called Canakit, explained what  had done to diagnose the problem. They agreed that the power supply was faulty, and shipped me a new one under warranty. They did not ask for the faulty one back, so I just hooked it up to a Pi 3B+ today with an SD card running OctoPi 0.15.1. It booted, but running much of a load, the lightning bolt came on. Shutdown and reconnected to one of my other supplies, and it boots from the same SD card and runs fine under load with no lightning bolt.

I like Canakit, and appreciate that they stand behind their products. However, in my (admitedly limited) sample, they are running a 25% fault rate on their included power supply.

LexBroski, 2019-01-15:
When I purchased my P3B+ back in March I had the exact same problem as John-Mc.  Faulty power supply.  Switched it out with a new one and haven't had a problem since.

tedder, 2019-01-18:
The undervolt warning in octoprint is simply calling the OS method that asks the CPU if it is being undervolted. If the OS is, or thinks it is, it **throttles your cpu**.

Unless evidence is provided otherwise, this is actually legitimate- in other words, many power supplies don't provide enough power to the Pi. I've actually designed some power supplies and set them to 5.2v or so, which is within range and ensures > 5.0v is received by the Pi, even under load. But long USB cables are a problem, especially if you are using a 'cell phone charger' power supply with a modern Pi3.

guysoft, 2019-01-30:
Reply from Canakit

Hello Guy,
Thanks for reaching out to us about the warning!

We warn our users that when using the Pi 3 B+ to make sure to be using Pi 3 B+ compatible operating systems, but we have not discussed anything about OctoPi.

If a user is using an older operating system that does not have Pi 3 B+ support, they will actually experience a misleading lightning bolt icon with a rainbow splash screen.  These are the symptoms that occur with the use of essentially any OS that did not have Pi 3 B+ support, such as any version of Raspbian that was available for download prior to March 14 2018 for example – regardless of the power supply.  

This pitfall commonly occurs if a user is taking an SD card they have prepared in the past from a Pi 3 B board onto their newly purchased Pi 3 B+.

Therefore, we send them a follow up email to make sure that they ensure use of Pi 3 B+ compatible operating systems.  The actual text is below for your reference.

IMPORTANT NOTES:

1. If you see lightning bolt symbol with a rainbow splash screen upon boot it indicates that you are using an operating system that is NOT compatible with the "B+" model. It is NOT related to the power supply voltage.

2. The "B+" model requires "B+" compatible software to operate (such as NOOBS 2.7.0 or later). Please ensure you are using a compatible operating system or the board will not boot.

I hope this helps clarify but please do not hesitate to contact us for any further information.

Many thanks,
Mike