guysoft/OctoPi #569: [Request] Adding 'change timezone' as an optional step to the how-to guide.

JohnEdwa, 2019-01-24:
#### Version of OctoPi: OctoPrint 1.3.10 running on OctoPi 0.15.1

Installing Octopi defaults the timezone to UTC +0 which usually isn't that important for using Octoprint, except if you install any plugins that deal with time like 'Display ETA'.   
As we are already asked to SSH to the pi to change the password, might as well add in `sudo dpkg-reconfigure tzdata` so it's done if the correct time is ever needed by the user.

JohnEdwa, 2019-01-24:
I mean exactly because it isn't configurable currently, adding it as a step to the "
How to use it?" guide in the readme would be helpful as we are already SSH'd to the pi:

> 4. Log into your Pi via SSH (it is located at octopi.local...
> 5. Set your timezone by running the ``sudo dpkg-reconfigure tzdata`` command.
> 6. To install plugins, pip may be found at...

(or whatever way is the simplest, that's the one I used)

Should obviously be added to the steps in https://octoprint.org/download/ as well.

guysoft, 2019-01-24:
1. You can install plugins via OctoPrints GUI.
2. I think those instructions belong in OctoPrint's FAQ at https://discourse.octoprint.org/c/support/faq because its user instructions and not build instructions, and they are not something you must do for OctoPi to run correctly. Suggest you put usage instructions there. @foosel can show you how to contribute there.

commit 6854edff (2019-02-11): Instructions how to optionally change timezone

Also relocated the "how to install plugins from the command line" step into the informational section since it's not a mandatory setup step considering that OctoPrint has a built-in plugin manager that works out of the box without needing to use the command line.

Closes #569