guysoft/OctoPi #570: octoprint does not connect automatilc on boot

sbias, 2019-01-24:
Hi,
(i hope this is not a dup, couldn't find this problem here)
while booting OctoPi, octoprint starts up. I set up automatic connect to the printer. But is most cases it does not work and i have to connect manual - which works fine.
After some debugging i found out, the process ModemManager is blocking the device while octoprint starts up. removing that package ( apt remove modemmanager ) solved it.
My recommendation: remove it by default. I think its not needed in octopi :)

sbias, 2019-05-05:
Hmm, just installed a new image of octopi, seems modemmanager has been removed already? can somebody confirm this? If correct, this issue can be closed.

guysoft, 2019-05-06:
modemmanager is not shipped with OctoPi, this issue can be closed.
[image]

closed 2019-05-06 by guysoft