guysoft/OctoPi #571: OctoPi 0.16.0 RC1 Status

guysoft, 2019-01-24:
Please try the release candidate.
Download it at:
https://storage.googleapis.com/octoprint/2019-01-24_2018-11-13-octopi-stretch-lite-0.16.0.zip

Md5: [``547babcd648f78e2d29e089dfd7e3538``](https://storage.googleapis.com/octoprint/2019-01-24_2018-11-13-octopi-stretch-lite-0.16.0.zip.md5).

### Changes in the image

* Raspberrypi 3A+ support
* linux kernel 4.14.79-v7+ from new base image 2018-11-13-octopi-stretch-lite
* OctoPrint 1.3.10
* Bug fixes

### Build notes
* Uses [CustomPiOS 1.1.0](https://github.com/guysoft/CustomPiOS/releases/tag/1.1.0)
* Our nightly/release builds have moved to the new Docker build supported in CustomPiOS 1.1.0. You can still use the vagrant build method.

tdub415, 2019-01-25:
Gonna throw this on a pi zero w for my monoprice mini :)

Anything in particular I should watch for?

Deneteus, 2019-01-26:
Ready to test!

Deneteus, 2019-01-26:
I tried upgrading an existing install to the RC by following this FAQ and nothing happened.

https://discourse.octoprint.org/t/how-to-use-the-release-channels-to-help-test-release-candidates/402

tdub415, 2019-01-26:
That's for octoprint, for octopi you'll need to flash the image to a SD card

ShohninDmitriy, 2019-01-26:
On the 0.16.0 version for about 5 months, there have been no problems not identified . The images he collected from from sources.

Deneteus, 2019-01-26:
> That's for octoprint, for octopi you'll need to flash the image to a SD card

I already had a couple cards I had upgraded the day before this post so I will just use those. I looked at the file info and they were the same. Is there an FAQ for how to upgrade without re-flashing?

guysoft, 2019-01-26:
You have to reflash to upgrade,  OctoPi is an image so there is no way
around that.
However OctoPrint can upgrade itself.

reverendbacon, 2019-01-26:
deploying today, thanks for the development

Mohoganogan, 2019-01-27:
Should "PIP" be upgraded? shows version 9 is installed but version 19 is available.
[image]

guysoft, 2019-01-27:
At the moment pip 9.0.3 is hardcoded in to the build because of #526

@foosel I guss since we are on 1.9.10 we can lift this?

https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/start_chroot_script#L40

tedder, 2019-01-27:
wrong screenshot? heh

ShohninDmitriy, 2019-01-27:
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.

John-Mc, 2019-01-27:
Gina has announced that OctoPrint 1.4 will be Python 3.

guysoft, 2019-01-29:
Ok, so there was a bug in the CI, and RC1 was only built against commit https://github.com/guysoft/OctoPi/commit/fb7093bfb950dc890323a4607068e8ceb5fa36cd

Issue was fixed and now will build against latest master. Expect RC2

voy13k, 2019-01-29:
Hello, Raspberrypi 3A+ with ender 3, stock firmware:
autoconnect on server startup option is ticked, but does not seem o autoconnect. Option tick is preserved across server or system restarts, but no effect. After manually connecting all works so far.

guysoft, 2019-01-30:
Closing, use #577

closed 2019-01-30 by guysoft

guysoft, 2019-01-30:
@voy13k Did that work in 0.15.1 for you? Is this s regression or just auto does not work?