guysoft/OctoPi #573: Serial port connectivity on rPI 3 A+ and Prusa MK3 not working

enyim, 2019-01-25:
#### What were you doing?

1. I've hooked up the 3 A+ to the MK# according to the manual at https://help.prusa3d.com/article/Loz15FAgEk-octo-print-raspberry-pi-zero-w
2. Connected to /dev/ttyAMA0
3. OctoPi did not connect

```
Connecting to: /dev/ttyAMA0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6b238af0, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
```

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPi

OctoPrint 1.3.10 running on OctoPi 0.16.0

#### Printer model & used firmware incl. version

Prusa MK3 fw 3.5.1
MMU 2.0 fw 1.0.3

### Toubleshooting

Tried on USB, it works.

Tried serial with a Pi Zero W, it works.

Tried various combinations of `enable_uart`, disabling BT, /dev/serialX, /dev/ttyS0, etc., it does not work.

Pressing connect then quickly resetting the printer prints the printer's boot log in OctoPi's terminal, and it progresses a couple of messages further after the initial N0 (it's random, how much), but the handshake never finishes.

Lino77, 2019-01-27:
works fine with me
MK3 firmware 3.5.1 RASPI 3a +

Lino77, 2019-01-27:
Connecting to: /dev/ttyAMA0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6f8efd70, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Prusa-Firmware 3.5.1 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Narda i3 MK3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok

guysoft, 2019-01-27:
cc @foosel ^

guysoft, 2019-01-31:
@enyim Thanks! your donation was received :1st_place_medal: 
Will test it today

guysoft, 2019-01-31:
Er,
I am not sure what I am supposed to test this, I just realised what pursa are doing is hooking the pi serial directly to the board, which I dont have.

BTW I can confirm that adding
```
enable_uart=1
```
At least works, so the serial connector seems to be behaving. 
Both my printers use RAMPS (RAMSP 1.4. and MKS gen_l)

cc @bubnikv have any insight on this?

Lino77, 2019-02-12:
what I noticed, the 3a + boots faster than the prusa. have Prusa MK3 with mmu2, until the prusa is up the octopi already has a mishap that I always manuel connect

guysoft, 2019-02-14:
@Lino77 Interesting.  Does that mean that once you connect once the Pursa MK3 boots up it works?

also cc @foosel , might need to change behaviour as Pis boot faster.

Lino77, 2019-02-14:
@guysoft , @foosel 
yes after connecting manually everything works fine. I have been printing for days.
I think that the MK3 takes too long until everything, including MMU2 is initiated. I Think a delay, or a small loop of two, three connection attempts, would probably solve the problem

guysoft, 2019-03-11:
@foosel Should I close this? Is this something you are planning to fix on your end?

foosel, 2019-03-11:
Not going to fix anything here. I can't add delays for everyone with every printer out there just because one model with one addon takes too long to initiate. If people run into this they can write a plugin or increase timeouts.

guysoft, 2019-03-11:
Closing as won't fix then.

closed 2019-03-11 by guysoft