guysoft/OctoPi #575: Not possible to reboot the system (Command for core:reboot failed with return code 139)

DominikPalo, 2019-01-28:
#### What were you doing?
1. Open the "Reboot/restart" menu from the topbar
2. Click the "Reboot system" item

#### What did you expect to happen?
The Raspberry Pi is restarted

#### What happened instead?
Nothing. But I see the error in the log:
```
2019-01-28 08:39:37,428 - octoprint.server.api.system - INFO - Performing command for core:reboot: sudo shutdown -r now
2019-01-28 08:39:37,486 - octoprint.server.api.system - WARNING - Command for core:reboot failed with return code 139:
! STDOUT: 
! STDERR: Segmentation fault
```

#### Did the same happen when running OctoPrint in safe mode?
I'm not even able to restart the system to the safe node (when I click "Restart OctoPrint in safe mode" nothing happens)

#### Version of OctoPi
OctoPrint 1.3.10 running on OctoPi 0.15.1

I have read the FAQ.

guysoft, 2019-01-28:
can you provide the output of 
```
dmesg | tail -n 200
```
?

DominikPalo, 2019-01-28:
It returns: `-bash: /usr/bin/tail: cannot execute binary file: Exec format error`, so I've executed it without `tail`:
```
...
...
...
[658289.781960] Under-voltage detected! (0x00050005)
[658296.021979] Voltage normalised (0x00000000)
[658506.102373] Under-voltage detected! (0x00050005)
[658512.342352] Voltage normalised (0x00000000)
[658516.502370] Under-voltage detected! (0x00050005)
[658520.662324] Voltage normalised (0x00000000)
[658537.302400] Under-voltage detected! (0x00050005)
[658558.102426] Voltage normalised (0x00000000)
[658770.262803] Under-voltage detected! (0x00050005)
[658776.502802] Voltage normalised (0x00000000)
[658953.303133] Under-voltage detected! (0x00050005)
[658965.783154] Voltage normalised (0x00000000)
[658972.023186] Under-voltage detected! (0x00050005)
[658976.183169] Voltage normalised (0x00000000)
[659356.823839] Under-voltage detected! (0x00050005)
[659363.063847] Voltage normalised (0x00000000)
...
...
...
```

DominikPalo, 2019-01-28:
Thanks, I know about the power issue, but it's a little bit strange - I tried three different power supplies (two 2.5A, one 2A) and that Raspberry Pi still reports the "Under-voltage" issue (although I measured the input voltage with a multimeter and there is 5.1-5.3V all the time). The same power supplies work fine on my other Raspberry Pies, so it seems there is some hardware issue on that particular Raspberry Pi - but otherwise, the RPi works completely fine for months (except the reboot issue, that I first encountered yesterday).

But a not-working tail command surprised also me, so I will try to run some scan/repair on my SD card and if it won't work, I will reinstall it, thanks.

Thanks, I know about the power issue, but it's a little bit strange - I tried three different power supplies (two 2.5A, one 2A) and that Raspberry Pi still reports the "Under-voltage" issue (although I measured the input voltage with a multimeter and there is 5.1-5.3V all the time). The same power supplies work fine on my other Raspberry Pies, so it seems there is some hardware issue on that particular Raspberry Pi - but otherwise, the RPi works completely fine for months (except the reboot issue, that I first encountered yesterday).

But a not-working tail command surprised also me, so I will try to run some scan/repair on my SD card and if it won't help, I will reinstall it, thanks.

tedder, 2019-01-28:
The pi3 is much more sensitive to undervoltage. Use shorter USB cables or (better) a PSU with an integrated cable, which is designed to compensate for the voltage drop.