guysoft/OctoPi #577: OctoPi 0.16.0 RC1 Status

guysoft, 2019-01-30:
Second release candidate after CI issue, has pip version 19 and not 9.

Please try the release candidate.
Download it at:
https://storage.googleapis.com/octoprint/2019-01-30_2018-11-13-octopi-stretch-lite-0.16.0.zip

Md5: [``c363d50f80f04bebcdc774ec20ad22ce``](https://storage.googleapis.com/octoprint/2019-01-30_2018-11-13-octopi-stretch-lite-0.16.0.zip.md5).

### Changes in the image

* Raspberrypi 3A+ support
* linux kernel 4.14.79-v7+ from new base image 2018-11-13-octopi-stretch-lite
* OctoPrint 1.3.10
* Bug fixes

### Build notes
* Uses [CustomPiOS 1.1.0](https://github.com/guysoft/CustomPiOS/releases/tag/1.1.0)
* Our nightly/release builds have moved to the new Docker build supported in CustomPiOS 1.1.0. You can still use the vagrant build method.

John-Mc, 2019-01-31:
Downloaded and installed. When going through the setup Wizard, it would not let me import a slicing profile for the Cura plugin. I got the error message:

> Profile import failed. Could not import the provided profile, it might be invalid. Check octoprint.log for details

I tried several different profiles, but got the same error message on all of them. These are the same slicing profiles I used successfully on OctoPi 0.14 and 0.15. I skipped this step in the Wizard, and the rest of the installation went normally.

After completing the Wizard, I went back to Settings> CuraEngine, and was able to install the slicing profiles without issue.

In attempting to download the log, I inadvertently deleted it. If it's worth recreating the incident, let me know and I will erase the SD card and start over.

John-Mc, 2019-01-31:
Thought I had deleted the log, but apparently I downloaded it first, so here it is. I'm 95% sure this was RC2, since I made burned the SD card from my computer, and I'm pretty sure I had deleted RC1 prior to making the card. Not sure how to tell, once the image is made.
[octoprint.log](https://github.com/guysoft/OctoPi/files/2817700/octoprint.log)

guysoft, 2019-01-31:
I can see two 403 errors:
```
2019-01-30 21:04:16,518 - tornado.access - WARNING - 403 POST /plugin/cura/import (fe80::4ad:e397:8801:670a) 49.74ms
2019-01-30 21:06:05,511 - tornado.access - WARNING - 403 POST /plugin/cura/import (fe80::4ad:e397:8801:670a) 42.23ms
```
cc @foosel

John-Mc, 2019-01-31:
>   You can check the version of pip:

This shows pip 19.0.1

> cat /etc/octopi_commit

Returns nothing. It appears to be an empty file.

guysoft, 2019-01-31:
Yes you are right, looks like /etc/octopi_commit is broken.
Comes from this commit: https://github.com/guysoft/OctoPi/commit/881d4165b2f85d62f863534420f93d80eccd397c
And would not work in the current vagrant/docker build methods. 
cc @foosel
It would require a change in the build system and I am not sure how to do it at the moment. Might consider dropping it.

drphil3d, 2019-02-01:
hey are you guys leaving ssh enabled by default with default root pi/raspberry credentials? you should at least change the ssh password so it's not super easy for anyone to just log into the thing on a network and have their way with it. You could easily burn somebody's house down with avrdude and some malicious firmware. avrdude isn't going to take the blame for it, hes was just there when it happened you can't prove anything. :)

foosel, 2019-02-01:
@drphil3d the setup instructions instruct the user to change the password. It wouldn't really make a ton of sense to simply set *another* default password that then is documented publicly. And SSH needs to be enabled because OctoPi is supposed to run headless.

guysoft, 2019-02-01:
We could use something like this in later versions:
https://github.com/jasbur/RaspiWiFi

drphil3d, 2019-02-01:
@foosel Yeah you're right that wouldn't help at all. The problem is a large majority of users don't even know how to use ssh and will simply ignore instructions to disable ssh not knowing the risk it poses. It's easy enough to add a blank ssh file to the boot directory to enable ssh if needed, or better yet how about an option to enable it through the setup wizard, but leave it disabled by default. 

@guysoft I worked on a project where we had to be able to configure the wireless network creds of a headless raspberry pi. We wrote a script that simply enables wifi connect after a button is pressed on the outside of the case. It's very similar to RaspiWifi just more refined. It creates a wifi hotspot and a captive portal. 
https://github.com/balena-io/wifi-connect

John-Mc, 2019-02-02:
Is the WiFi hotspot option discussed in #396 included in this release? If so, I'd like to test it out, but not sure where to find it.

guysoft, 2019-02-02:
No, no hotspot was added. I think we might try stuff out in 0.17.0.

foosel, 2019-02-05:
> It's easy enough to add a blank ssh file to the boot directory to enable ssh if needed

That is far from easy for a lot of people. Editing an existing text file that contains instructions on how to edit it is already challenging, creating an empty one without a file extension? Forget about it. I've had to deal with support requests for this for several years now, the common skill threshold is way longer than you'd think and mildly infuriating at times.

> or better yet how about an option to enable it through the setup wizard, but leave it disabled by default.

The most common cases where SSH is needed is if the UI is no longer working for some reason and it isn't an option anymore to enable SSH through it. And enabling it just in case as part of initial setup doesn't change anything compared to simply having it enabled from the get go. So either you haven't gained anything, or you end up locked out when stuff goes wrong.

The problem here isn't that SSH is running on the Pi by default but that people don't read instructions that tell them to change the bloody password, and other than having OctoPrint refuse to run until that is done (which would in turn cause issues because OctoPrint is tailored to run on a ton of setups, not just OctoPi) you cannot force people to do it. I also don't want any kind of OctoPrint plugin on OctoPi that allows to change the password or do other administrative stuff (or even offer console access) because that opens a whole can of worms I really don't want opened considering that users are happily making their instances' port 80 publicly accessible on the internet (thankfully they don't appear to do that a lot with port 22 since that's not needed for regular operation, only when stuff goes really wrong). 

Disabling SSH by default would add a severe barrier to supporting users - and quite frankly I'm already at my absolute limit regarding handling the amount of support requests that come in and the only reason I haven't yet declared defeat is the community forum where people now at least help each other. It runs headless, it needs port 22 or the only option when it breaks for some reason for most people will be to start over, and guess who'll get their head ripped off for that?

guysoft, 2019-02-05:
Will also note this debate is not relevant for this release candidate. 
During a release candidate we freeze all new features unless there is a critical issue or escalation. 

If you want a new feature pushed or discussed, please open a separate issue.

guysoft, 2019-02-11:
Ok, 12 days, no critical bugs, no escalation, this RC is the new 0.16.0

guysoft, 2019-02-11:
Released:
https://github.com/guysoft/OctoPi/releases/tag/0.16.0

If you guys wanna start hotspot testing fun then continue it at #396 , this is the time to test and have fun and breaks stuff.

closed 2019-02-11 by guysoft

Deneteus, 2019-02-16:
> Released:
> https://github.com/guysoft/OctoPi/releases/tag/0.16.0
> 
> If you guys wanna start hotspot testing fun then continue it at #396 , this is the time to test and have fun and breaks stuff.

Why doesn't the date on the Octo pi image reflect when you made the RC2 version live?  It's confusing to unzip an image to see an older date for what is supposed to be a newer release. I get that you put the  raspbian release date out front but you can get that info from the release notes, and the command line.

I would expect any new Octopi image date to match the date released.

[image]

guysoft, 2019-02-17:
@Deneteus That is the date the base image of Raspbian was released.
You can see the list of versions here: https://downloads.raspberrypi.org/raspbian_lite/images/