guysoft/OctoPi #579: Raspberry Pi 3 Model A+ problem wont find internet using 0.16.0

BEARCUBB, 2019-02-05:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

John-Mc, 2019-02-05:
Have you tried working through the [WiFi Setup and Troubleshooting Guide](url=https://discourse.octoprint.org/t/wifi-setup-and-troubleshooting/184)?

BEARCUBB, 2019-02-05:
> Have you tried working through the [WiFi Setup and Troubleshooting Guide](https://discourse.octoprint.org/t/wifi-setup-and-troubleshooting/184)?

Yes went back through it all found one thing about the State im in and also where it was asking for my name in the IMG file SSID name i placed my routers name "Not what it wanted" i placed the pi name and it found it all works for now. Thanks for the link and help.

closed 2019-02-05 by guysoft