guysoft/OctoPi #580: Octopi.local won't open in Chrome Browser

rockman44181000, 2019-02-13:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

tedder, 2019-02-13:
Zero information?

[Read this](https://community.octoprint.org/t/octopi-wifi-network-connection-troubleshooting-megatopic/2608). Ask questions there. But you'll need to supply much more information.

rockman44181000, 2019-02-13:
I was working on the newest Beta Version of OctoPrint and all of a sudden I can't open it in any of my Browsers. I have tried Brave, Mozilla, Chrome, Edge and Internet Explorer. It was working perfectly until today. I can connect through Putty without the slightest issue. I've tried Port Forwarding using DemoForges Mirage Driver with TightVNC and a whole myriad of other options to no avail. I was trying today to setup Printoid, when everything went wonky. Now I would just like to actually print something and keep an eye on it, but I can't load anything from my desktop to my Octoprint Server without my Desktop connecting. I do have Bonjour installed also.

rockman44181000, 2019-02-13:
Information such as? I've gone through the entire OctoPi WiFi/network connection troubleshooting megatopic. And I can't find anything that is even close to the issue that I am having. I can't get into the octopi.log files to even have a chance to supply the information that is supposedly required.  As I said before, I can open up Putty and login, yet I can't login to any browser as they all say that the connection has been refused. I am routing through a Hitron CGN3RON Router and I've gone through each of the settings and the Router can see my Octopi either through 2.7GHZ. Ethernet or through the 5GHZ. Wireless connection. It is allowed through the Mcafee Firewall and I have Bonjour Print Server 2.02 installed. Other than that without being able to get into the Octopi.log files. I don't know what you or anyone else needs. I'm not that Network Savvy. I can usually get by, except in this issue. I've even reset my Router to Factory Defaults all to no avail.

guysoft, 2019-02-13:
1. if you can ssh to your Pi using putty then this is not a network issue anymore, because the IP is resolved by your machine
2. does putting the IP address of the Pi in your browser help?
3. You said you are using a beta version of OctoPrint, are you sure the Octoprint server is up and running?
What is the output of
```
ps aux | grep octoprint
```
From your ssh (putty) connection?

tedder, 2019-02-13:
You'll note I said "zero information" because you opened the ticket with .. zero information. You only added it later- after my comment.

Does Octoprint load in the browser if you access it by IP, rather than `octopi.local`?

rockman44181000, 2019-02-13:
After using ps aux | grep octoprint from the ssh Putty connection I came up with the following:
pi 448 6.5 5.6 310876 53 684 S1 15:41 0:27 home/pi/octoprint/bin/python2 /home.pi/octoprint/bin/octoprint serve --host= 127.0.0.1 --port=5000
pi 1527 0.0.0.0 4372 536 pts/0 S+ 15:48 0:00 grep=color=auto octoprint

guysoft, 2019-02-14:
Ok, and does it OctoPrint work using the IP and not ``octopi.local``? Both me and @rockman44181000  asked you because that tells us if this is hostname resolve issue.

rockman44181000, 2019-02-14:
No, I realize it is a Hostname Issue. It's just that I don't know how to go about correcting it. I just had my service provider perform a Factory Reset and though I can get into the Raspberry Pi B3+ through Putty. I can't get into OctoPrint itself through any of my Browsers for some reason. It keeps telling me that the Octopi is refusing the connection.

tedder, 2019-02-14:
`octopi.local` comes from zeroconf/bonjour. You are on windows, here are some solutions:
* [FAQ: I can't reach my OctoPi under octopi.local ](https://community.octoprint.org/t/i-cant-reach-my-octopi-under-octopi-local/210)
* [workarounds](https://github.com/OutsourcedGuru/octoprint-name-resolution-hacks)

rockman44181000, 2019-02-14:
I tried installing Samba onto my Pi and though I succeeded, it looked vastly different tan what was listed in the accompanying file. I can only get so far and then I get a Loading Error notice when it finishes loading. I may have to give up and just hookup a laptop to my printer and connect it to my network to transfer files from my Server.

guysoft, 2019-06-27:
This is becoming a configuration issue and not an OctoPi issue, please use the community forum for that:
http://discourse.octoprint.org/

closed 2019-06-27 by guysoft