guysoft/OctoPi #581: [Request] Consider updating /etc/sudoers.d/* scripts as created

OutsourcedGuru, 2019-02-15:
#### What were you doing?
I was trying to copy/paste the existing `/etc/sudoers.d/octoprint-shutdown` script to create my own. It resulted in a situation in which the system simply ignored my efforts, forcing the `pi` user to still include the password on `sudo` attempts.

From the OctoPi script which creates it:
`echo "pi ALL=NOPASSWD: /sbin/shutdown" > /etc/sudoers.d/octoprint-shutdown`

I found that the following updated style of syntax seems to work as expected for adding the `uhubctl` executable:
`echo "pi ALL=(ALL) NOPASSWD: /sbin/shutdown" > /etc/sudoers.d/octoprint-shutdown`

Admittedly, the Debian man pages for this are fairly terse.

#### What did you expect to happen?
I expected my own copy/paste attempt to create `/etc/sudoers.d/uhubctl` from your script would magically allow me to run this executable from within an OctoPrint plugin.

#### What happened instead?
Since the attempt at adjusting `sudoers` failed, behind-the-scenes OctoPrint couldn't invoke the executable and failed with a `subprocess.call()` exception.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPi
1.3.10

#### Printer model & used firmware incl. version
Virtual Printer

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2019-02-19:
You need to make sure the permission of the files are right. 
See: https://askubuntu.com/questions/159007/how-do-i-run-specific-sudo-commands-without-a-password

What is the output of 
```
ls -l `/etc/sudoers.d/uhubctl`
```
Also in the example you pasted you didnt change ``/sbin/shutdown"``, what is the binary?

OutsourcedGuru, 2019-02-20:
Here is the rights as seen in the folder so it looks like mine's the same as yours in this case:

```
-r--r----- 1 root root  28 May  7  2018 010_pi-nopasswd
-rw-r--r-- 1 root root  35 May  7  2018 octoprint-service
-rw-r--r-- 1 root root  32 May  7  2018 octoprint-shutdown
-r--r----- 1 root root 958 Jun  5  2017 README
-rw-r--r-- 1 root root  90 Feb 15 09:17 uhubctl
```

What I'm suggesting is the syntax I had to use was "ALL=(ALL)" rather than the version you used.

Worked:

```
pi ALL=(ALL) NOPASSWD: /home/pi/oprint/lib/python2.7/site-packages/usbcontrol/bin/uhubctl
```

Didn't work:

```
pi ALL=NOPASSWD: /home/pi/oprint/lib/python2.7/site-packages/usbcontrol/bin/uhubctl
```

guysoft, 2019-02-20:
What version of OctoPi are you using? you specified the version of OctoPrint in that field.

OutsourcedGuru, 2019-02-20:
Ah (sorry). That would be 0.15.1.

mvp, 2019-03-06:
Note that you do not have to use sudo for uhubctl if you configure udev properly, see more here: https://github.com/mvp/uhubctl#linux-usb-permissions

OutsourcedGuru, 2019-03-07:
Remember, this is a plugin so it's not on my computer that the change is taking place; it's on everybody's who installs this plugin (OctoPrint-USBControl). The current strategy is that the Settings panel prompts the user to go run a single command at a command prompt... which strikes me as slightly easier in this case. Duly-noted, though.

guysoft, 2019-04-23:
Can this be closed or there anything else needed?
(closing if no reply)

OutsourcedGuru, 2019-04-25:
Not sure what to tell you. I can only guess that the `shutdown` command within a classic OctoPi-imaged installation can successfully do a password-less `sudo shutdown` from the System menu. And yet, it feels like the syntax could be wrong per the specs.

Feel free to close. Just remember this for later if it ever breaks.

guysoft, 2019-05-01:
Roger

closed 2019-05-01 by guysoft