guysoft/OctoPi #584: Plymouth is prevented from running by hard-coded text startup script.

WheresWaldo, 2019-04-07:
#### I want a boot logo

Attempting to install a boot screen using Plymouth. The approach chosen was to minimize other software running on the rPi and keep the image as small as possible. Plymouth is installed by default in rasbian Stretch lite and it should be as simple as creating a theme in /usr/share/plymouth/themes, for example mytheme.plymouth in its own aptly named directory. Then setting it as default.

1. make necessary modifications to /boot/config.txt and /boot/cmdline.txt to force the rPi to boot quietly
2. create a theme with appropriate .plymouth files
3. set theme as default with sudo plymouth-set-default-theme
4. sudo reboot

#### I expected the boot logo png file to display

#### The default text based script displays instead

At the end of the text display, I do get a messages about the Plymouth Boot screen, they are as follows:

Starting Terminate Plymouth Boot Screen...
Starting Hold until boot process finishes up... 

#### Did the same happen when running OctoPrint in safe mode?

<OctoPrint is not the issue, the issue is the way this distribution is configured. OctoPrint runs fine in all instances so no need to test in SAFE mode.

#### Version of OctoPi

Current released version 0.16.

#### Printer model & used firmware incl. version

Printer model is not applicable as this is a distribution setup issue

[config.txt](https://github.com/guysoft/OctoPi/files/3051685/config.txt)
[cmdline.txt](https://github.com/guysoft/OctoPi/files/3051686/cmdline.txt)
[theme.zip](https://github.com/guysoft/OctoPi/files/3051689/theme.zip)

I have read the FAQ.

WheresWaldo, 2019-04-07:
By the way, If I take stock Raspbian Stretch and execute most (cherry pick parts) of your build script, or build my own OctoPrint instance using the OctoPrint wiki as a guide, plymouth works as expected.

WheresWaldo, 2019-04-16:
Adding some more background, I was able to get Plymouth working, but the added text still displays on the rPi. I disabled the MOTD service as well as edited PAM so that nothing goes to SDOUT. I deleted the welcome script also. Since my Raspbian mojo is not 100%, I am at a loss for what keeps the messages appearing on the screen over the Plymouth theme.

guysoft, 2019-04-17:
Ok, I am not really sure what I can do on my end for this. If you want to provide steps I could add a CustomPiOS module that adds a Plymouth with a theme.

WheresWaldo, 2019-04-17:
The issue is that your custom text seem to run no matter what is done with Raspbian. Plymouth has been patched and now is completely compatible with the rPi distro, including your version. But what I really want is a way to not execute your welcome script.

WheresWaldo, 2019-04-17:
I am going to close this issue. I found where the welcome script was being run from, ~/.bashrc

closed 2019-04-17 by WheresWaldo