guysoft/OctoPi #585: Issues with latest 0.16 build and Logitech C270 Camera

MrJabu, 2019-04-20:
i think there is a issue with the latest build. i could not get the c270 working for the life of me. i messed with default settings and changing octopi.txt & config.txt parameters. nothing worked.

i then loaded the 0.15 image and everything worked right away without having to change anything.

guysoft, 2019-04-20:
Output of ``dmesg`` command when you plug it in? Might get more info from
the driver.

guysoft, 2019-04-23:
Might be related to: https://github.com/guysoft/OctoPi/issues/499

podmore71, 2019-07-11:
Having the same issue recently. Sometimes it works, other times it doesn't.  at first I was using a ps3 webcam and thought that might be the problem so I bought the C270, but it's still doing it. Here is the output of the 'dmesg'

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.42-v7+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1219 SMP Tue May 14 21:20:58 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x37800000
[    0.000000] On node 0 totalpages: 229376
[    0.000000]   Normal zone: 2016 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 229376 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 227360
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=d86f492f-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 886308K/917504K available (8192K kernel code, 627K rwdata, 2164K rodata, 1024K init, 821K bss, 23004K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xb8800000 - 0xff800000   (1136 MB)
                   lowmem  : 0x80000000 - 0xb8000000   ( 896 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 628 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26229 entries in 77 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000287] Console: colour dummy device 80x30
[    0.000907] console [tty1] enabled
[    0.000963] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001008] pid_max: default: 32768 minimum: 301
[    0.001364] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001401] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002331] CPU: Testing write buffer coherency: ok
[    0.002806] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003479] Setting up static identity map for 0x100000 - 0x10003c
[    0.003645] rcu: Hierarchical SRCU implementation.
[    0.004454] smp: Bringing up secondary CPUs ...
[    0.005314] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006241] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007103] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007223] smp: Brought up 1 node, 4 CPUs
[    0.007304] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007328] CPU: All CPU(s) started in HYP mode.
[    0.007348] CPU: Virtualization extensions available.
[    0.008336] devtmpfs: initialized
[    0.020127] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.020396] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.020444] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.021048] pinctrl core: initialized pinctrl subsystem
[    0.021938] NET: Registered protocol family 16
[    0.024915] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.030659] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.030696] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.030929] Serial: AMBA PL011 UART driver
[    0.033201] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[... 230 lines cut ...]
[    9.029968] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.031597] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    9.031611] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[... cut ...]
[   16.317158] Bluetooth: L2CAP socket layer initialized
[   16.317179] Bluetooth: SCO socket layer initialized
[   16.331908] Bluetooth: HCI UART driver ver 2.3
[   16.331916] Bluetooth: HCI UART protocol H4 registered
[   16.331960] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.332073] Bluetooth: HCI UART protocol Broadcom registered
[   16.532875] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.532882] Bluetooth: BNEP filters: protocol multicast
[   16.532895] Bluetooth: BNEP socket layer initialized

guysoft, 2019-07-11:
Related:
https://community.octoprint.org/t/issue-with-rpi-3-and-logitech-c270-camera/9091

This is the relevant output from your dmesg:
```
[ 3.475822] videodev: Linux video capture interface: v2.00
[ 3.554474] uvcvideo: Found UVC 1.00 device (046d:0825)
[ 3.563820] sd 0:0:0:0: Attached scsi generic sg0 type 0
[ 3.675093] uvcvideo 1-1.1.3:1.0: Entity type for entity Extension 4 was not initialized!
[ 3.675111] uvcvideo 1-1.1.3:1.0: Entity type for entity Extension 6 was not initialized!
[ 3.675123] uvcvideo 1-1.1.3:1.0: Entity type for entity Extension 7 was not initialized!
[ 3.675135] uvcvideo 1-1.1.3:1.0: Entity type for entity Processing 2 was not initialized!
[ 3.675147] uvcvideo 1-1.1.3:1.0: Entity type for entity Extension 3 was not initialized!
[ 3.675159] uvcvideo 1-1.1.3:1.0: Entity type for entity Camera 1 was not initialized!
[ 3.675567] input: UVC Camera (046d:0825) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/input/input0
[ 3.675877] usbcore: registered new interface driver uvcvideo
[ 3.675884] USB Video Class driver (1.1.1)
```

Looks like  its detected,  Do you have any ``/dev/video0`` file?

podmore71, 2019-07-11:
Thanks for the quick reply. There is a file named video0 in the /dev folder

Andy

podmore71, 2019-07-11:
Just had a quick power cut and on restart it's working again but don't know how long it will last

podmore71, 2019-07-11:
So I came back to it this afternoon and low and behold, it did it again. I'm running Octopi ver 0.16 and Octoprint 1.3.11. I even tried installing 0.15.1 with 1.3.11 and it still did it, although it is currently working now. It's a very strange and intermittent problem

guysoft, 2020-10-02:
So is this fixed? Can I close this?

closed 2020-10-02 by MrJabu