guysoft/OctoPi #588: Raspberry Pi Camera V2.1 module won't work with Octoprint

SeventhDread, 2019-05-04:
I have plugged in my camera won't come up I have tried everything I have connect the 15 pin cable in the right way I am completely new to this so please answer as simply as you can I have tried all sorts of things so please talk me through step by step (With pictures if you can) like from the very start with a blank micro SD, Raspberry Pi 3 B and a Raspberry Pi Camera Module V2.1 please I really want this to work

tedder, 2019-05-04:
Seventh, there are plenty of walkthroughs. Google it and find one that fits your learning style.

John-Mc, 2019-05-04:
This site is focused mainly on bug reports, not on providing “how-to” assistance. 

You might try searching (or posting) on https://community.octoprint.org/. That site is focused more on providing support, including for set-up issues such as yours.

guysoft, 2019-05-05:
Thanks @tedder and @John-Mc for pointing @SeventhDread to the right place in my absence. Closing, this is not a bug, a community question for https://community.octoprint.org/

closed 2019-05-05 by guysoft