guysoft/OctoPi #589: Is it just me or is it a Bug?? Downloading Octoprint through Raspbian

SeventhDread, 2019-05-05:
I am following theese directions https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337 and when I do the first big bit of code in terminal in Raspbian is sats Do you want to continue [Y/n] I keep saying Y and it says Abort why is that is there a different way to download Octoprint on Raspbian so I can control it on Raspbian? Or is this the right thing but it is just a bug? I am using a Raspberry Pi 3 or 3B please help I just want to control my printer on a Raspberry pi that I can hang on the wall and use it for Octoprint and other stuff thanks.

John-Mc, 2019-05-05:
> org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337 and

The easiest way to get up and running is by going to https://octoprint.org/.
Scroll down and click the download button. Follow directions there to
download and install OctoPi. This is an IMG file that includes the
Raspian-lite OS along with Octoprint. You can install on an SD card, make
some minor set-up tweaks, and be ready to go.

Installing over an existing installation of Raspian is possible, but more
complicated to get up and running.

guysoft, 2019-05-06:
Also, its not supported here :)
This is the OctoPi issue tracker, not Octoprint's. And this is not an issue, so you probably want http://community.octoprint.org/

Or just install OctoPi, its pretty neat ;)

closed 2019-05-06 by guysoft