guysoft/OctoPi #591: [Request] Add Pillow package

FormerLurker, 2019-05-12:
Is it possible to add the Pillow package to the next version of OctoPi? Lots of users have trouble installing this package, usually due to PSU problems, but it often causes the install for my plugin (Octolapse) to fail.

Thanks for your consideration!

#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2019-05-12:
PSU problems mean you have other issues, not software one.
Make sure you using a 2.5A power supply and a good USB cable.

I dont want to increase the footprint of the distro for software only a fraction of the users are going to download and use. It does not make sense.

Again, it sounds like what you want is a decent installation script.

FormerLurker, 2019-05-13:
@guysoft, Pillow is installed via PIP within setup.py, just like every other requirement as is suggested within the standard setup.py template provided for plugin development.  Pillow is a very common module and can be found in many common and popular plugins.  @foosel doesn't want any plugins that run shell commands within the plugin repository (for good reason, obviously), so if there is a problem installing the entire plugin install will fail, and you can imagine how difficult these issue are to debug.

I wouldn't have asked for this had I not exhausted every other option.  Also, I have seen this exact issue (which is not always hardware related, fyi) within other popular plugins.  Since this module is THE most popular image manipulation library, and image manipulation is such a common requirement, and the install is error prone, I believe it warrants further consideration.

guysoft, 2019-05-15:
You can also build your own variant, something like this:
https://github.com/guysoft/OctoPi/blob/pillow/src/variants/pillow/post_chroot_script

Then you can add what ever you want to the image.

FormerLurker, 2019-05-16:
Thanks for that!  I'm not familiar with how/when those scripts are executed, but I'll take a look and see if I can't figure it out.

guysoft, 2019-05-19:
@FormerLurker Instructions in the readme:
https://github.com/guysoft/OctoPi/#building-octopi-variants

guysoft, 2019-06-27:
Did you get any progress?
I am sort of inclined to close this distro, and encourage you to build a variant.

guysoft, 2019-07-02:
Closing due to no response.

closed 2019-07-02 by guysoft