guysoft/OctoPi #592: [Request] Add gphoto2

FormerLurker, 2019-05-12:
Would it be possible for you to pre-install the latest stable version of GPhoto2 on the next OctoPi release?  It can be a tricky package for users to install, and needs to be added to the sudoers file in order to make it useful for users.

GPhoto2 is useful in my plugin (and I'm guessing several others) for capturing images from DSLR cameras.  [Here](https://github.com/gonzalo/gphoto2-updater) is a link to a repo that contains a very convenient script that can be used to install the latest version of gphoto2 from github.  

The script there can be executed like this:
```
 sudo gphoto2-updater.sh -r
```
Here is a sudoers entry that I believe is safe to use:
```
# allow 'sudo gphoto2' to run without supplying a password
pi ALL = (root) NOPASSWD: /usr/bin/gphoto2
```

Thank you very much for your consideration!
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2019-05-12:
I am not shipping a huge blown image viewer to OctoPi, which is by default, it has no relevance AFAIK to other users and its has a massive footprint on the distro.
Furthermore you did not supply any source code to your plugin.

You are more than welcome to use the variant system to build your own distro.

closed 2019-05-12 by guysoft

FormerLurker, 2019-05-13:
@guysoft,

I understand why you might want to limit the size of your distro, but you are mistaken about the source code for my plugin.  It is available [here](https://github.com/FormerLurker/Octolapse/).

FormerLurker, 2019-05-13:
> It would be possible though, it you want to maintain an install script, to have that in the scripts folder. So the process would be easier.

I'm currently trying to communicate with the author of the updater script I originally posted about inclusion within the OctoPi scripts folder.  It is open source, but I want to get explicit permission to use and modify this script before I submit another feature request.  I appreciate your flexibility here, as including this script would cut down on the manual installation process very substantially, and would mostly resolve one of the most common and difficult to debug issues I receive.

I will not continue to post to this closed issue anymore.  Thank you for your time.