guysoft/OctoPi #593: [Request] Allow loopback address (127.0.0.1) to access mjpegstreamer's control.htm file

FormerLurker, 2019-05-12:
Since default access to /webcam/control.htm has been removed due to security concerns, I've had some difficulty with my plugin (Octolapse).  Specifically changing webcam settings.  Users are required to make changes to octopi.txt to allow access to edit webcam settings through my plugin.  However, my plugin only requires access to the address locally.  It would be safer (and easier for my users) to allow access to http://127.0.0.1/webcam/control.htm by default than to open up all of those pages via a settings change in octopi.txt, right?

So, my request is this:  Would it be possible enable the loopback address to access the control.htm page for mjpegstreamer?

#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPi

#### Printer model & used firmware incl. version

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

guysoft, 2019-05-12:
I am not sure if there are security implications on this, @foosel any input?

foosel, 2019-05-13:
Might be possible to add some sort of ACL to haproxy to allow that, but I'm not sure. We are currently disallowing commands generally using the `-n` flag. So any `?action=command` on the streamer endpoint gets ignored. One would need to see if it's possible to craft a haproxy rule that allows these kind of queries from localhost but not anywhere else. If so we could drop `-n`.

guysoft, 2019-07-02:
Closing if no reply

FormerLurker, 2019-07-03:
If you are still willing to consider this, id appreciate it if you left this one open.  I don't know much about haproxy yet unfortunately, so I am unable to provide any specific implementation unfortunately.

guysoft, 2019-07-03:
OK