guysoft/OctoPi #594: Support for a hardware button and indicator for shutting down Octopi on-location?

bwaslo, 2019-05-17:
I have a nagging issue with Octopi in that its SD card can get (and has gotten) corrupted unless I first do a shut-down of the program before removing DC power from the Pi.  Octopi is great for letting me print and monitor without having a computer other than Pi in the room with the printer.  But I STILL have to get to a computer or use my phone (if I remember to bring it with me!!) to 'shut-down' and then get back to the printer to power down!  Not convenient, particularly when I am making changes to the printer and something goes wrong requiring me to unpower for fixing it.

Would it be possible to add a feature utilizing a couple of the Raspi's I/O pins:

- one for a button (hold down for a few seconds to activate) and 

- the other for an indicator LED (to indicate when shutdown is complete and it is safe to remove power from the Pi)?

I wouldn't mind trying to program this myself, but my skills are sadly dated (Windows programming using Delphi) and I have only minimal experience with Linux or C.  If this feature can't be added to the distro, perhaps someone could point me to tools or tutorials for modifying the code for this?

Thanks
... and...
I have read the FAQ.

guysoft, 2019-05-19:
What version OctoPi are you using?
Because I haven't had corruption on shutdown in a while here.

guysoft, 2019-05-19:
User resolved his own issue, comment here:
https://guysoft.wordpress.com/2013/08/26/octopi-0-6/#comment-8420

Url linked in comment:
https://www.stderr.nl/Blog/Hardware/RaspberryPi/PowerButton.html

closed 2019-05-19 by guysoft