guysoft/OctoPi #595: ssh keys broken

andrewschott, 2019-06-09:
OK, a fresh download and install of OctoPi, logged into it physically, updated, did the usual any competent linux admin would do, rebooted.

This is where the fun began.  

SSHd didnt work.  Well it did, but looking at "journalctl -xe"'s output, it hated every key.  All along I thought it was me being a moron, but apparently the ssh keys are bogus that come along for the ride.  Once I regenerated them, things worked out ok.

```
# ssh-keygen -f /etc/ssh/ssh_host_rsa_key -N '' -t rsa
# ssh-keygen -f /etc/ssh/ssh_host_dsa_key -N '' -t dsa
```

tedder, 2019-06-10:
What are the errors?

guysoft, 2019-06-10:
1. What version of OctoPi?
2. what is the output of the log: ``/var/log/regenerate_ssh_host_keys.log``?
(Similar case, but only for first boot https://github.com/guysoft/OctoPi/issues/424#issuecomment-340811799)

guysoft, 2019-06-17:
Closing if no reply

andrewschott, 2019-06-17:
Sorry was out of town.  Just was able to ssh in and see if I had any logs and this is the best I can provide offhand.  Need to dig more.  As to your original question, was the latest available on the octopi download page, immediately updated (had monitor and keyboard physically attached to pi).  After updated, confirmed that OctoPrint worked, I connected it to the printer.  At this point, I realized I forgot to enable the camera and found that I could not SSH in.  Physical keyboard attached I went thru things and realized it was the keys, regenned them, and problems gone.

```
May 31 22:34:41 octopi systemd[1]: Starting OpenBSD Secure Shell server...
May 31 22:34:41 octopi sshd[21936]: key_load_public: invalid format
May 31 22:34:41 octopi sshd[21936]: Could not load host key: /etc/ssh/ssh_host_rsa_key
May 31 22:34:41 octopi sshd[21936]: key_load_public: invalid format
May 31 22:34:41 octopi sshd[21936]: Could not load host key: /etc/ssh/ssh_host_ecdsa_key
May 31 22:34:41 octopi sshd[21936]: key_load_public: invalid format
May 31 22:34:41 octopi sshd[21936]: Could not load host key: /etc/ssh/ssh_host_ed25519_key
May 31 22:34:41 octopi sshd[21939]: key_load_public: invalid format
May 31 22:34:41 octopi sshd[21939]: Could not load host key: /etc/ssh/ssh_host_rsa_key
May 31 22:34:41 octopi sshd[21939]: key_load_public: invalid format
May 31 22:34:41 octopi sshd[21939]: Could not load host key: /etc/ssh/ssh_host_ecdsa_key
May 31 22:34:41 octopi sshd[21939]: key_load_public: invalid format
May 31 22:34:41 octopi sshd[21939]: Could not load host key: /etc/ssh/ssh_host_ed25519_key
```

guysoft, 2019-06-18:
Hum, how about the date of that file?
``ls -l /var/log/regenerate_ssh_host_keys.log``

Might tell us if it was regenerated after you initial install.
But I really not sure what to ask for and figure out what went wrong.

andrewschott, 2019-06-18:
Doesn't look like it.

```
------------------------------------------------------------------------------
OctoPrint version : 1.3.11
OctoPi version    : 0.16.0
------------------------------------------------------------------------------

pi@octopi:~ $ sudo -s
[sudo] password for pi: 
root@octopi:/home/pi# ls -tl /var/log/regenerate_ssh_host_keys.log 
-rw-r--r-- 1 root root 61 Nov 13  2018 /var/log/regenerate_ssh_host_keys.log
root@octopi:/home/pi# 
```

guysoft, 2019-06-20:
I am not really sure what log or information at this point would let us figure out what has went wrong

ggamble22, 2019-11-25:
I just wanted to leave a fix for anyone who runs into this issue like I did. You just need to regenerate your keys using the following commands:

```
sudo ssh-keygen -f /etc/ssh/ssh_host_rsa_key -b 2048 -t rsa
sudo ssh-keygen -f /etc/ssh/ssh_host_ed25519_key -b 521 -t ed25519
sudo ssh-keygen -f /etc/ssh/ssh_host_ecdsa_key -b 521 -t ecdsa
```

I was able to get away with just the first line but this gets rid of all the error.

tedder, 2019-11-25:
That isn't a fix, it's a cargo-culted workaround. Calling it a fix implies you know what the problem was.

guysoft, 2019-11-26:
@tedder @ggamble22 There should be a fix in place, if you are getting the bug then it means that something is not working in the fix. Please provide ``/var/log/regenerate_ssh_host_keys.log`` If its there. That should give us more information why you are not getting ssh keys.

ggamble22, 2019-11-29:
@tedder Fair enough. I will update my comment. I just wanted to help anyone running into this issue.

ggamble22, 2019-11-29:
@guysoft My /var/log/regenerate_ssh_host_keys.log just contains: 
`ssh-keygen: generating new host keys: RSA ^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@`

Let me know if there is anything else I could do to help.

timhughes, 2020-11-07:
> I just wanted to leave a cargo-culted workaround for anyone who runs into this issue like I did. You just need to regenerate your keys by running the following commands on the pi:
> 
> ```
> sudo ssh-keygen -f /etc/ssh/ssh_host_rsa_key -b 2048 -t rsa
> sudo ssh-keygen -f /etc/ssh/ssh_host_ed25519_key -b 521 -t ed25519
> sudo ssh-keygen -f /etc/ssh/ssh_host_ecdsa_key -b 521 -t ecdsa
> ```
> 
> I was able to get away with just the first line but this gets rid of all the error.

You should be able to do the following to replace them all correctly
```
root@raspberrypi:~# rm -f /etc/ssh/ssh_host_*
root@raspberrypi:~# ssh-keygen -A
ssh-keygen: generating new host keys: RSA DSA ECDSA ED25519 
```

guysoft, 2020-11-08:
Wee have code that handles this now, and no complained the past year. so closing this unless someone says otherwise.

closed 2020-11-08 by guysoft